christianhaitian/arkos #523: Feature Request: show screenshot on different save slots

qianhd, 2022-06-25:
Hello, Christian,
Hope you are doing well!

There is 10 save slots in emulators, but i have no idea about the detail of these save points, unless i load them one by one.
from technical point of view, is it possible ?
if yes, where is the relevant code ?

christianhaitian, 2022-06-25:
Since the majority of the emulators used in ArkOS are based on retroarch, you would have to look there.  Not something I would look to implement but you’re welcome to check with libretro on future consideration for this.

closed 2022-06-25 by christianhaitian

qianhd, 2022-06-25:
> Since the majority of the emulators used in ArkOS are based on retroarch, you would have to look there. Not something I would look to implement but you’re welcome to check with libretro on future consideration for this.

thank you for your information.