christianhaitian/arkos #527: Yabasanshiro 1.9 Issues

agorsar, 2022-07-02:
I'm running the latest ArkOS on my rg351mp, and loaded some games to check the improvements with Yabanshiro, the Sega Saturn emulator.
I tried Sonic 3D Blast, Nights into Dreams, Shining Force III and Legend of Oasis. Every game had flickering issues not present on the other emulators/cores. For example, Nights has flickering on text, while Legend of Oasis has the flickering on shadow effects.
Is this emulator specific and thus happening on other distributions too, or is this an ArkOS specific problem?

christianhaitian, 2022-07-02:
I haven't had this issue on my unit.  Try standalone-nobios or standalone-bios and see if there's a difference there.

christianhaitian, 2022-07-03:
I checked this again and I do notice some flickering of text for Night into Dreams when entering your name.  Gameplay seems ok.  That's the emulator and not something specific to any OS.  Hopefully it will improve in a future update of the Emulator.

closed 2022-07-03 by christianhaitian