christianhaitian/arkos #531: Wifi hangs the 351MP device after joining network.

molcos77, 2022-07-21:
On the latest available Arkos (01.07.2022) installed fresh on a 351MP device i noticed that sometimes the system hangs when i input the Network dongle. If i have a network configured on the device the process of attaching the network dongle hangs the device.
If i delete the network connection on the device and then attach the network dongle, it works and the 351mp continue to work as before.

If i then try and connect to the network, it hangs the device after typing in the wifi passcode.
I have tried with a different USB-C to USB-A dongle with the same result. I also tried with the network dongle that came with my RG351P device, but that dongle was not recognized by the 351mp.

I am not sure if it has something to do with the SOC at all, so i am now charging the device to a 100% to see if the problem persists.

Regards

molcos77, 2022-07-21:
I think it might be related to the battery level, i charged it for an hour and tested when the device was at 95% (Allegedly)
And then it worked fine.

I don't know if there is a fix for this but this is what my amateurish testing concluded.

christianhaitian, 2022-07-22:
It's mostly battery related.  The battery measure on the 351 devices are wildly in accurate. So if it's at around 60 percent or less, it will start to act erratic like that.  Could even be around 70 percent as well.  Good luck.

closed 2022-07-22 by christianhaitian