christianhaitian/arkos #533: Running X11 application

mgord9518, 2022-07-26:
I've started working on a game in the Ebiten engine, which runs just fine on a Raspberry Pi and I'm trying to port it over to my RG351P running ArkOS. After some debugging, I see that my game is crashing due to not being able to find an X11 display.

How does one go about running an X11 app on ArkOS? My game isn't missing any libraries and was compiled on Ubuntu 18.04 so that's not the issue. I've already tried setting DISPLAY to :0 and startx; it doesn't appear to exist in the system though.

christianhaitian, 2022-07-26:
X11 video drivers are not available by default in ArkOS.  Just GBM.  If you want X11, you'll have to install X11 g31 video drivers for the rk3326 chipset. https://github.com/JustEnoughLinuxOS/libmali/tree/main/lib/aarch64-linux-gnu

mgord9518, 2022-07-26:
@christianhaitian Thanks for the reply, is GBM just something I'd have to build against instead of X? I'm pretty inexperienced with the embedded side of things

Or would connecting the device to the internet, SSHing in and installing X11 be the easier route?

christianhaitian, 2022-07-26:
X11 may work but I've never tested it with this chipset.  GBM is what most of the distros make use of on this chipset so if you can building against GBM, it would be better I think.

christianhaitian, 2022-07-27:
I saw the authors post about the GBM.  You can try pointing your binary towards that x11 driver and see if it works.  Might be the solution for now for your use case provided it works.

closed 2022-07-30 by christianhaitian

tinspin, 2024-09-17:
What is the difference between libmali-bifrost-g52-g2p0-x11-gbm.so and libmali-bifrost-g52-g2p0-gbm.so/libmali-bifrost-g52-g2p0-x11.so?

How would one go about to launch a binary towards another driver?