christianhaitian/arkos #534: Portmaster not working/starting on OGA 1.1

harshitgoel96, 2022-08-09:
### Describe the Issue (If applicable)
When I try to start port master, I get "testing for sudo", and the terminal screen with blinking cursor. 

### How can the issue be reproduced? (If applicable)
Updated ArkOS, Options > Tools > portmaster > portmaster

### What device are you using?
Odroid GO Advanced Black Edition (1.1)

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
ArkOS 2.0

### Anything else you'd like to include that may help to review this issue or feature request?
This is first time I am trying to start portmaster and install moonlight.

christianhaitian, 2022-08-09:
Try moving the portmaster.sh script out of the portmaster folder and into the roms/tools folder and run it from there.  As for your pairing issue in Moonlight, did you make sure to put in the correct pc name or ip for your device in the menu?

harshitgoel96, 2022-08-12:
> Try moving the portmaster.sh script out of the portmaster folder and into the roms/tools folder and run it from there. As for your pairing issue in Moonlight, did you make sure to put in the correct pc name or ip for your device in the menu?

Moonlight on OGA1.1 does not let me enter the pc detail. it directly gives error could not connect to pc

harshitgoel96, 2022-08-12:
some more test I did, Moonlight works fine on the RetroOz OS. so its not a device related issue. and wifi is also not the issue. 
will try with a fresh install of arkos on another SD card once more

harshitgoel96, 2022-08-12:
Hi, I have the portmaster and moonlight now working on OGA 1.1. the main difference I see is the build version. I have 07022022 in the firmware where portmaster and moonlight both are working. While the original SD card where I have issue is on 070320220 build. I thinking something in the OTA update messed up the portmaster system?

christianhaitian, 2022-08-12:
No updates in the 07302022 update would have affected moonlight or portmaster.  Both work fine on my end with the 07302022 build.

harshitgoel96, 2022-08-14:
Ok, I got it working on 07032022 build as well. I think there was a script failure when I tried to install port master for the first time, and after that it did not start. But after a fresh install and update to 07032022. Both portmaster and moonlight work

christianhaitian, 2022-08-14:
Great.  Happy you got it figures out.

closed 2022-08-14 by harshitgoel96