christianhaitian/arkos #535: File Manager onscreen keyboard not showing properly - RG351MP

Akira-N28, 2022-08-11:
Hi,

whenever I open up the on-screen keyboard in the File Manager (for example when I need to name a new folder or rename a file), the keyboard appears "squished" : the keys are not clearly visible and are somewhat overlapping (see attached picture), and the footer exceeds screen borders. Is there a way to fix this?

I'm on Anbernic RG351MP, running ArkOS 2.0 (07012022).

[image]

christianhaitian, 2022-08-12:
I'll provide a fix for that in a future update.  Until then, use 351Files as it's a better file manager designed for these rk3326 devices.

christianhaitian, 2023-01-21:
I have not been able to resolve this keyboard issue.  The source used for it is [here](https://github.com/christianhaitian/rs97-commander-sdl2) if someone would like to tackle it.  Otherwise, 351Files is a much better file manager in my opinion.

closed 2023-01-21 by christianhaitian