christianhaitian/arkos #537: Possible bug: Tools menu option disappears after switching to SD2 card

borf42, 2022-08-19:
Hi ChristianHaitian,

I am working on setting up my devices, and I was having trouble getting to PortMaster. If I do a fresh install of ArkOS, it will show up under Options then Tools. But when I switch to SD2 for ROMS, that tool menu goes away. I have tried this on both a RG503 and RG351V. On both devices, that Tools menu disappears and won't come back, even if I switch it back to the main SD card. 

I hope I am clear on my description of the issue. If you need any log files or screen shots, please let me know.

Thank you so much for your work on this fantastic OS!!

christianhaitian, 2022-08-19:
Hello,

The tools menu goes away because it doesn't find any tools in the roms2/tools folder.  Reinstall something like ThemeMaster or PortMaster there and it will reappear again.

closed 2022-08-19 by christianhaitian

borf42, 2022-08-19:
Please forgive my ignorance, but I am not sure how to do reinstall ThemeMaster or PortMaster. If it isn't too much trouble, can you point me in the right direction?