christianhaitian/arkos #539: Duckstation on RG351MP

agorsar, 2022-08-24:
Hi Christian,

There seems to be a problem withDuckstation on my RG351MP. I'm using the latest update (23/08/2022), and Duckstation simply doesn't launch.
I choose a game, then Emulation Station launches the emulator, but it crashes back to Emulation Station immediately. If I press "A" (for per game configuration) yellow letters appear on the console indicating a config file has been created, then it crashes to the PSX menu.
Everything else seems to be working (gliden64 on the previous update seems to boost some n64 games!)

Thanks for your incredible support, as usual.

christianhaitian, 2022-08-24:
How many sd cards are you using?

agorsar, 2022-08-24:
I am using two SD cards.

christianhaitian, 2022-08-24:
Do switch to main sd card, then do switch to sd2 again and see if that fixes the issue for you.

agorsar, 2022-08-24:
Just a question before doing that. Will it somehow change any setting I might have?

christianhaitian, 2022-08-24:
Only if you manually changed locations of roms for emulators in the es_systems.cfg file in the /etc/emulationstation folder.

agorsar, 2022-08-24:
Everything is working.
Is the duckstation settings reachable with any key combination? Performance seems inferior as is compared to the default retroarch core (testing Tekken 3).

agorsar, 2022-08-24:
Oh I just checked the key combination in the changelog!
Thanks.

closed 2022-08-24 by agorsar

christianhaitian, 2022-08-24:
Yea the emulator may not be the best option for this chipset but in some cases, it may be preferred.  It works better on the rg503/rg353p given they're better chipsets.  You can try changing the renderer to from OpenGL to software to see if that helps.  Otherwise, maybe tinkering with some of the other menu options.

agorsar, 2022-08-24:
And as a note, There's the option to use Vulkan (which is obviously unsupported). If you change to Vulkan on the general settings, you won't be able to enter the emulator anymore as it crashes. Onyl going into linux and changing the ini file does the trick (not accessible by mounting the sd card on Windows).
Yes, I did it by accident lol

christianhaitian, 2022-08-24:
Interesting.  Good to know.  I'll see if we can build it and make Vulkan unavailable.

Majjio, 2022-09-03:
> Interesting. Good to know. I'll see if we can build it and make Vulkan unavailable.

I made that mistake lol I updated the firmware & was intrigued to try the Duckstation standalone emulator & before I even tried a game i fiddled with the settings & set Vulkan, now can't enter any game, how do I edit that file the person mentioned above? If it's too complicated I probably will stick to using one of the other cores, my mistake, I'm using the Chi device if that's any help.

Majjio, 2022-09-03:
I managed to find the file & edit it 🙂 I got an app on my phone a file manager one I connected to the device & edited the file Vulkan to Open GL & it worked. Also not related but i appreciate the retroarch tate core, there's a shader in the CRT folder called mame something & there's a few good ones in that folder I like e.g mame scanlines. 
[image]

christianhaitian, 2022-09-03:
The next update will include an updated Duckstation binary that prohibits that Vulkan setting from causing that issue again.