christianhaitian/arkos #540: Option to Disable Boot Rumble

0xLinkz, 2022-08-25:
Saw an old post about this from 2021 about this, but Anbernic does have a solution for this, can we have an option for this?

JoeLyga, 2023-01-21:
Out of curiosity, which model has the boot rumble?  My RG351V with the stock firmware does this, but I haven't had boot rumble on any other device or firmware besides that one.

christianhaitian, 2023-01-21:
351 p/m/v only and only with stock and ArkOS.  AmberElec disabled rumble on boot in their firmware.  I never got around to disabling rumble on boot as I wanted to incorporate the ability for the user to pick having rumble enabled at boot or not.  I may not get to that.

Toufailleur, 2023-07-21:
Hi @christianhaitian, do you have any update? Could we set the parameter ourselves on the firmware? Thanks for your hard work!

christianhaitian, 2023-07-22:
Not yet.

christianhaitian, 2023-10-17:
I've decided that this is not something I will no longer pursue as I no longer have a rg351v unit and I've moved on to rk3566 devices.  I will continue to support the rg351v for new emulator and updates and such.

closed 2023-10-17 by christianhaitian