christianhaitian/arkos #541: Portmaster is not installing ports

waeggy123, 2022-08-29:
Device: RG351V 
Latest ArkOS 2.0 release installed -> Portmaster is  installing ports without any problems
(Same with ArkOS Update 07/30/2022)

ArkOS Update to 08/22/2022 -> Portmaster showed message when trying to install a port:
Files can not installed please check Internet connection

christianhaitian, 2022-08-30:
Are you sure your device is connected to the internet when you try to install ports?  If so, try reinstalling portmaster.

waeggy123, 2022-08-30:
Yes, the device is connected to the internet, even checking again the ArkOS Update function is working and some Ports are downloading, but right before installing it stops with this message.

Reinstalling Portmaster (Replacing folder and .sh file) showed the same error message. 

This process was repeated six times with different SD cards and two different RG351V devices as well with 3x fresh downloaded img file (ArkOS 2.0 for RG351V) and two different internet network and the error could be reproduced.

christianhaitian, 2022-08-31:
This issue has now been resolved with the latest update to PortMaster.  Just update through the PortMaster menu or download the latest PortMaster.zip file from the PortMaster location.  Thanks for reporting the issue.

closed 2022-08-31 by christianhaitian