christianhaitian/arkos #543: RG503: amiga standalone (amiberry) audio is at max volume and cannot be controlled with volume rocker

mjkaye, 2022-08-31:
On the RG503, when the amiga core is set to `STANDALONE` (amiberry), games run with the sound at maximum volume. The volume cannot be reduced by using the volume rocker. The only way the volume can be controlled is to bring up the menu (with `SELECT + X`) and move the `Paula Volume` slider in the `Sound` sub-menu. The `Paula Volume` slider has to be adjusted and saved for each game individually.

Tested on update 08222022.

On an RG351MP, running update 11052021, the volume rocker successfully controls the sound volume under the same circumstances.

christianhaitian, 2022-08-31:
This is a known issue and there’s no known fix for it at this time for the rg503.  You may find that one of the retroarch Amiga emulators give similar performance and have working volume control.

mjkaye, 2022-08-31:
Thanks for the quick reply, Christian. You're right; performance with retroarch/puae is fine, so I'll stick with that for now and close this issue.

closed 2022-08-31 by mjkaye