christianhaitian/arkos #544: Battery level stays at 100% on OGA 1.0

klundry, 2022-09-01:
I have just gotten ArkOS loaded and working on my Odroid Go Advance 1.0 version handheld. The battery level is stuck at 100 and never changes. If I put my sd card in with a different image the battery level reports correctly so it's not a hardware issue.

Anything I can try?

christianhaitian, 2022-09-01:
That's because the rk2020 that I target that's the same as the oga 1.0 doesn't have the power management chip like the oga 1.0 does.  So I don't use the pmic logic that's available for the oga 1.0.  I can provide a dtb with the pmic logic added for use with the oga 1.0.  May take me a few days to get to it.

christianhaitian, 2022-09-02:
Attached, is the updated kernel and dtb that should report battery level more correctly for the oga 1.0.  Just copy them over the existing ones in the fat32 boot partition and test and confirm that it resolves the issue.
[KernelnDTBforOGA1_0.zip](https://github.com/christianhaitian/arkos/files/9474212/KernelnDTBforOGA1_0.zip)

klundry, 2022-09-02:
Yes, it is reporting the correct battery level now. Thank you so much!