christianhaitian/arkos #545: REQUEST - I've been trying to load a raw ubuntu instalation on my RG351P, but i dont know where to start

Leandro-Reis, 2022-09-02:
I was wondering if the Arkos team could release a tutorial on how to install Ubuntu on a RG351p or even a version of ArkOs without all the emulators and tools. I've been searching for information on how to do that on the internet but haven't found anything. I want to use my RG351p as a development station, to install PHP, mysql, git etc. Is actually possible to do it with the arkos regular version but I wanna know how to boot my own version of Ubuntu on my console.

I'm not intended to install a UI or anything like that, a raw version of Ubuntu running on the console and accessible by ssh would do. Of course, if it's too much work for the team, you can disregard this request. 

Ty.

christianhaitian, 2022-09-02:
Hi Leandro,

Have you reviewed the building info here yet? https://github.com/christianhaitian/arkos/wiki/Building
If so, let me know where you're getting stuck at.

christianhaitian, 2022-09-02:
Also, I don't recommend doing alot of building on the rg351.  It's not a vented unit so excessive heat will not properly escape and will wear out components.  I recommend doing these builds on a VM.  The link above has more info about this VM.

closed 2022-09-10 by christianhaitian