christianhaitian/arkos #548: RG351v 3DO does not appear in main menu ui

varkanoid, 2022-09-04:
RG351v V2.0 (01/07/2022 release to v2.0 (08/22/2022)

The 3DO system is missing from the main menu of emulation systems.

The ROM folder name is 3do so this is created when the ROM partition is created/expanded (!?)
The es-systems.cfg.txt shows the folder name as 3DO (see below)
If I rename the folder to 3DO then it appears in the main menu screen and fixes the issue.
Small change/issue but thought its worth bringing to your attention.

<system>
		<name>3DO</name>
		<fullname>3DO</fullname>
		<path>/roms/3DO/</path>

christianhaitian, 2022-09-04:
exfat is case insensitive and doesn't need this.  Are you using a 2 SD card setup and your second sd card is setup with ext4?

varkanoid, 2022-09-04:
[image]

Single 400gb SD card as exfat. If I rename it back to 3do it disappears from the menu, rename it back to 3DO and it appears again.

varkanoid, 2022-09-04:
Also noticed another issue N64DD is also missing from the main menu of emulators. I have a ROM folder called n64dd, I can see it in the es_systems.cfg.txt file but doesnt show up on main menu or under the view emulators settings page ?

christianhaitian, 2022-09-04:
The folders won't show up in the menu if they are not populated with at least one file with a supported extension.

christianhaitian, 2022-09-05:
I never tried N64DD so didn't know honestly.  That will most likely need updating.  Can you test and verify that they do work with that emulator?  I wonder if the standalone can read N64DD as well.

varkanoid, 2022-09-05:
Mugen ones dont work they all say "failed to load content" when loading the .n64dd ROM, ParaLLei is the same if I go into Retroarch/Retroarch32 and load core then load content. 

So I used Project64 on the PC to test they actually worked. Needed to download the japanese ipl N64 BIOS/rom as the games were all japanese versions but once did that they worked fine. I dont know how to test the standalone games as N64DD doesnt appear in the emulator menu or emulator settings so cant test them and dont know how to do it.

varkanoid, 2022-09-05:
Found this for N64dd

https://forums.libretro.com/t/is-there-a-mupen64plus-next-64dd-guide-anywhere/35014/6

varkanoid, 2022-09-05:
Blimey, I put the Japanese IPL.N64 file in \system\mugenplus64 directory like the above link said and now I've got an entry for N64DD !!

varkanoid, 2022-09-05:
just need to figure out why its not showing the game list just IPL (which appears to be the name of the BIOS file I copied in \system\mugenplus64 but getting close.

first need food.

[image]
[image]

varkanoid, 2022-09-06:
Just update to your latest version. Thanks very much. :)

closed 2022-09-10 by christianhaitian