christianhaitian/arkos #549: EasyRPG doesn't work on RG503 (crash at game startup)

LoSmarrito, 2022-09-05:
Hi, 
on my RG503, when I load any game compatible with EasyRpg, the game crashes after a couple of seconds (I just see the ArkOs banner twice, before going back to the game collection menu). The game scanning has run fine, I can see the .easyrpg files.

Latest version of the OS, games switched on the second sd card. (I've already tried to go back to the "first" one, but the issue remained). Folders in bios\rtp\2000 and bios\rtp\2003 look to be present and filled (automatically downloaded once I run the games)

The same games look to work fine on the same device in JelOS, so they shouldn't be corrupted.

Thank you for all your work! If you have any question, please tell me.

christianhaitian, 2022-09-05:
I tried 2 EasyRPG games on my end:

 - Final Fantasy IX Demake in a .easyrpg folder 
 - 8 bit Quest V1.0.2.zip

both started up without issue on my end with ArkOS (08222022) using a 2 sd card setup.  What specific EasyRPG games are you having trouble loading?

LoSmarrito, 2022-09-05:
Laxius Power I
Fifth Era - Fragments of the Holy Stone
Ib v1.07

LoSmarrito, 2022-09-05:
I' ve redownloaded Ib, deleting the old file, and it started fine... Don't know why, honestly.

I guess that doing the same for the other games will do the trick

I guess you can close this issue, christian, thank you for your time.

closed 2022-09-05 by christianhaitian