christianhaitian/arkos #551: Portmaster: Fade to Black not loading

waeggy123, 2022-09-10:
The Portmaster game "Fade to Black" (f2bgl) is not loading.

Log file shows:
Failed to create secure directory (/run/user/1002//pulse): No such file or directory
f2bgl: resource.cpp:168: void Resource::loadINI(File*, int): Assertion `levelsCount == kLevelDescriptionsCount' failed.

System:
ArkOS 2.0 (latest update 09052022)
Portmaster (latest version 6.56)
Fade to Black CD (Europe)

christianhaitian, 2022-09-14:
I loaded Fade to Black on my RG503 using ArkOS 2.0 (08052022) and it started just fine using the latest Portmaster version available (6.56).  My Fade to Black CD is USA based.  Perhaps you didn't copy the necessary files to the appropriate location or it may not be compatible with the Europe version? https://github.com/christianhaitian/PortMaster/wiki#f2bgl-fade-to-black

waeggy123, 2022-09-14:
Finally the US based CD version of Fade to Black was running.
(Europe and downsized US DOSBox versions are not compatible even when all files are present)

closed 2022-09-14 by waeggy123

christianhaitian, 2022-09-14:
Thank you for sharing.  I’ll add that to the portmaster wiki so others are aware.

waeggy123, 2022-10-11:
Thanks for the feedback and testing.
In this case it seems to be the problem with the european version, because all files were copied.

I checked it on other RG351 devices as well as on the RG552 and RG503 without any success.

Cheers,
Andreas ;)