christianhaitian/arkos #552: Gamecube emulator on RG503

timwillhack, 2022-09-15:
Apologize to put this here.  I have been unable to find anything related to a game cube emulator running on the rg503 until I ran across your emulator list.  However I am unsure how to go about setting it up.  Any suggestions?  I'm on windows.  Thank you

christianhaitian, 2022-09-15:
You must be on ArkOS version 09/05/2022 or later.  You then put the compatible game files in the gc folder on the sd card then it should show up in the emulationstation menu.  If the gc folder doesn't exist on the sd card, simply create it and add the files there.

timwillhack, 2022-09-15:
Ah, ok wow, I had just bought the device a week or so ago and just downloaded the arkos release but didn't try to update yet.  Thank you!  You have some great work!