christianhaitian/arkos #553: Factory reset

andreamorello93, 2022-09-16:
How can i do a Factory Reset for ArkOS 2.0? i have a RGB20S device

the menu messed up cause there isn't any confirmation of language changing and now has all DOTS and squares (i think symbols).

I would just to restore settings.

christianhaitian, 2022-09-16:
I don't support the RGB20S so I don't know how to help you much on this device.  You can try removing the line that says language in the /home/ark/.emulationstation/es_settings.cfg file and it should revert back to English.  Good luck.

closed 2022-09-16 by christianhaitian

andreamorello93, 2022-09-16:
ty...i have seen that for the RGB20S there is no download...but ArkOS is installed on this device. how can i re-flash it at it's version?

christianhaitian, 2022-09-16:
You'll have to contact the manufacturer.  The manufacturer installed their own version of ArkOS on it.  I have no clue what they did to it.