christianhaitian/arkos #554: ScummVM crashes Escape from Monkey Island

ridgekuhn, 2022-09-18:
Hey friend, sorry about this super half-baked bug report for a game with experimental support in ScummVM, lol. I think I'm just looking for advice!

Monkey Island 4 crashes ScummVM when trying to save/load a game using the OpenGL With Shaders renderer. It works fine with software rendering (but runs like trash), and it also works on my desktop, so I'm thinking the RK3326's OpenGL isn't compatible with the shader renderer? I suspect it might work with the OpenGL Classic renderer, but it seems to be missing from the ArkOS build. 

Anyway, ScummVM dies with no helpful error messages in the console, so I've tried rolling my own binary to test my theory, but I've been having a hard time getting it to compile and/or run. What config options are you using to build ScummVM?

christianhaitian, 2022-09-18:
Hey Hey!

Happy to hear from you again.  Hope you're doing well!  You can see how I build a majority of the core stuff in ArkOS for the rk3326 here, https://github.com/christianhaitian/rk3326_core_builds.  For scummvm standalone, the specific recipe I use is this, https://github.com/christianhaitian/rk3326_core_builds/blob/rk3326/scripts/scummvm.sh.  Let me know if a later build fixes the issue as I'll add it as part of a future update.

ridgekuhn, 2022-09-29:
@christianhaitian Yes, doing well, thanks so much for asking! This is great, I should have gone through your repos before asking, but this looks like exactly what I need. I'll let you know how it goes!

ridgekuhn, 2022-10-06:
@christianhaitian Looks like this [isn't possible with GLES1/2](https://github.com/scummvm/scummvm/blob/db56c9dab0ceef83e15b32025adf9f2712e33ef1/configure#L5613). Thanks for your help anyway, Christian!