christianhaitian/arkos #556: Building instructions: most RG503 links are broken

mjkaye, 2022-09-22:
In the "Building" document - https://github.com/christianhaitian/arkos/wiki/Building - the following links (related to the RG503) are broken:

- kernel; https://github.com/christianhaitian/RG503_kernel.git
- kernel building instructions; https://github.com/christianhaitian/RG503_kernel/blob/main/README.md
- u-boot; https://github.com/christianhaitian/RG503_uboot

The broken links are due to the fact that neither the https://github.com/christianhaitian/RG503_kernel nor https://github.com/christianhaitian/RG503_uboot repositories exist.

Would it be possible for those repositories to be uploaded to github? Is there a version of the RG503 building instructions that doesn't require those repositories?

Thanks.

christianhaitian, 2022-09-22:
Yea those are forks from Anbernic's repo and they still haven't made them public for some reason.  I've went ahead and copied the forks to new repositories and made them public so they're available now.  The links in the build instructions have been updated to reflect this change.

closed 2022-09-22 by christianhaitian

mjkaye, 2022-09-23:
Thank you, Christian. It's much appreciated.