christianhaitian/arkos #558: American Laserdisc Games

varkanoid, 2022-10-03:
RG351V
ArkOS 2.0 (09/05/2022)

Hi,

I finally figured out how to get American Laserdisc Games to work and what the issue is. Not sure whether you can do anything about it but I thought I would describe the issue. Its probably more to do with Emulationstation.

When you launch a game (with all the correct files in \ROMS\ALG you get a black screen and the "save me" wav is played then it goes back to the menu screen. If you look at the error below this is because the ALG folder does not exist in that path. Instead its singe/johnnyrock/cdrom-globals.singe as you can see on the attached screenshot at the bottom. All the .singe game files refer to \singe\ALG\Johnny....... which is where its getting the incorrect path from. The Singe folder is mapped from \ROMS\ALG as Singe it probably needs Singe\ALG creating. Anyway to prove this I copied the whole of the Johnnyrock folder to \ROMS\ALG\ALG so that when it came to find the game files CDROM, png etc it could find them. 

The log file shows below 👎 
2022-10-03 13:28:44.449 INFO  [1367] [reset_logfile@324] Version v2.8.2c-1-dirty
2022-10-03 13:28:44.450 INFO  [1367] [reset_logfile@328] Command line: /opt/hypseus-singe/hypseus-singe singe vldp -framefile /roms/alg/johnnyrock/johnnyrock.txt -script /roms/alg/johnnyrock/johnnyrock.singe -homedir /opt/hypseus-singe -datadir /opt/hypseus-singe -sound_buffer 2048 -volume_nonvldp 5 -volume_vldp 20 
2022-10-03 13:28:44.450 INFO  [1367] [reset_logfile@329] CPU : UnknownCPU || Mem : 896 megs
2022-10-03 13:28:44.459 INFO  [1367] [reset_logfile@330] OS : Linux 4.4 || Video : Anbernic
2022-10-03 13:28:44.472 INFO  [1367] [reset_logfile@331] RGB2YUV Function: C
2022-10-03 13:28:44.472 INFO  [1367] [reset_logfile@337] Line Blending Function: C
2022-10-03 13:28:44.472 INFO  [1367] [reset_logfile@344] Audio Mixing Function: C
2022-10-03 13:28:46.371 INFO  [1367] [ldp_vldp::read_frame_conversions@1043] Framefile parse succeeded. Video/Audio directory is: /roms/alg/johnnyrock/./
2022-10-03 13:28:46.399 INFO  [1367] [printline@96] Starting Singe version 1.50
2022-10-03 13:28:46.450 INFO  [1367] [printline@96] SINGE: Script Error!
2022-10-03 13:28:47.451 INFO  [1367] [printline@96] SINGE: error compiling script: cannot open **singe/_ALG_/johnnyrock/cdrom-globals.singe:** No such file or directory
2022-10-03 13:28:47.451 INFO  [1367] [printline@96] SINGE: Cannot continue, quitting...

[image]
[image]

So if you make it look like this and basically have two copies of the game folder in different areas it works.
[image]
[image]
[image]

christianhaitian, 2022-10-03:
I only tested 1 game with American Laser so you may have stumbled on to something here.  Can you try Ninja Hayate and see what you get with that?

varkanoid, 2022-10-03:
I have figured it out. There are two game versions floating about. One for Daphne-Singe and one for Hypseus-Singe. I was using the Daphne-Singe configured games hence why the directory's are different. I've just downloaded some configured for Hypseus-Singe including fan made and the two that were never released Platoon and Tron and they work now. I was just being a bit of a confused plank.

closed 2022-10-03 by varkanoid

christianhaitian, 2022-10-03:
No worries and this is good info you shared.  I'll make sure to update the wiki entry for American Laser Games with that info.  Thanks again

varkanoid, 2022-10-04:
No problem, the video is running to slow I might have a look at that later in the week when I`m meant to be working as Daphne video plays fine. Didnt realise that the scripting for the actual "game" is quite basic (well basic as I can understand it :) )and I can see why people have made their own games.