christianhaitian/arkos #560: Sound output resets when launching games

mattcoad, 2022-10-07:
Not sure if it's a bug or issue with Linux, if you have headphones plugged in and launch a game it outputs from both headphones and speaker.
If you unplug and replug the headphones the sound then correctly outputs only in headphones

Any way to fix this ?
Thanks

christianhaitian, 2022-10-07:
What unit and what version of ArkOS?

mattcoad, 2022-10-07:
Rg503 09/05

christianhaitian, 2022-10-07:
Did you reboot the device with the headphones plugged in before you launched a game?

mattcoad, 2022-10-07:
I didn't, because I didn't know that was a thing
I'll try that tomorrow

christianhaitian, 2022-10-07:
No it’s not something that should be needed.  Just trying to figure out what caused that.  I have a workaround for an issue related to headphone detect I will be making available in the next update and want to see if it might solve this issue as well.  What game were you playing when you noticed this issue?

mattcoad, 2022-10-07:
It was snes, pretty much any game I launched.
It also happened on mame, I was on a plane with my headphones on and a few angry people tapped me to tell me they could hear my game lol

christianhaitian, 2022-10-07:
Lol.  Ok, let me see if i can replicate the issue.

PurplePanda88, 2022-10-07:
Hey I know your super busy, but im too dumb to build this myself so I was wondering if you can add this port for the RPG Maker XP <333. it only seems to be 25 lines or so and would be sick for pokemon fan games. This dude figured it out last year so it would be so sick if you added it I just got my rg351p.

RPGMAKERXP for rg351- https://github.com/krishenriksen/AnberPorts/issues/51
if it helps here's a version for 1.8+ of Unbuntu
https://github.com/mkxp-z/mkxp-z/tree/v2.4.0-fix (v2.4.0-fix)

PurplePanda88, 2022-10-07:
I literally can't read it, its like a recipe in German. Also everyone would love it and retrogamecorp would prolly make a video. <3 much love christian

christianhaitian, 2022-10-07:
I couldn't replicate this issue on my rg503 with ArkOS 2.0 version 9/05/2022.  The only way I could get audio out of both the speaker and the headphones at the same time was if I didn't plug the headphones in all the way into the headphone jack.

christianhaitian, 2022-10-07:
or if I left the headphones plugged in between reboots then audio comes out of the headphones and the speakers at the same time until I unplug and replug the headphones again which I have a solution coming for that issue in the next update.  It would not change the output of audio between games though.  I wonder if between launching games if you were rebooting the device while the headphones were still in?

mattcoad, 2022-10-07:
When you say reboot do you mean full device reboot or just start + select to exit game?
I just did a video of the issue on my device.

christianhaitian, 2022-10-07:
full device reboot.

mattcoad, 2022-10-07:
Sorry video took forever to upload

https://filebin.net/fjv5y9hc94vkr7vr

christianhaitian, 2022-10-08:
hmmm...I can't get that to happen on my end at all.  Were there any changes you made to retroarch?

m235iau, 2022-10-08:
Nope literally out of the box config

m235iau, 2022-10-08:
I can try a clean reflash of the sd card?

christianhaitian, 2022-10-08:
Let me try a fresh flash on my end first

christianhaitian, 2022-10-08:
I've confirmed the issue on the fresh flash.  I have a resolve I will include with the next update.

m235iau, 2022-10-08:
You are a legend mate appreciate all you do!

christianhaitian, 2022-10-30:
This issue should be resolved after yesterday's update.

closed 2022-10-30 by christianhaitian