christianhaitian/arkos #563: Unable pair device with Moonlight

Orion81, 2022-10-15:
I use anche rg503 with latest version of ark OS.
I try to connect the device with moonlight but no way, every time it gives me the same error: unable to pair check name of PC ecc ecc.
I try with my chrome book on the same network and it works.

christianhaitian, 2022-10-15:
Did you connect your rg503 to your wireless network and confirm it communicates?

Orion81, 2022-10-15:
It gives me connected on menu. I don't have internet connection (i use hot spot mobile when i need) on this Lan because I use it only for local game streaming.

christianhaitian, 2022-10-16:
Can you ssh into the device?

christianhaitian, 2022-10-16:
Enable Remote services from the options section and try to ssh into it.

Orion81, 2022-10-16:
[image]

I don't Know how to ssh. This is what apend

Orion81, 2022-10-16:
Ok I make everything you write in question "How do I SSH into ArkOS?"

Now what can I do?

christianhaitian, 2022-10-16:
Can you access the device from your computer on the network?  If so, than the device is not the issue.  I have a new moonlight update coming soon that may fix this issue for you.

Orion81, 2022-10-16:
True Ark Browser? Yes It works!

christianhaitian, 2022-10-16:
Moonlight has been updated to the latest version.  Please reinstall it from PortMaster and try it again.  Let me know if it resolves the issue for you or not.

Orion81, 2022-10-16:
I installed it now. Same issue. My pc name is "casa". I've only need to put "casa" as name?

christianhaitian, 2022-10-16:
that should work if that's truly the name.  If not, try putting in the ip address of your pc.

Orion81, 2022-10-16:
How can I find the ip adress?  It Is a lan without internet connection

christianhaitian, 2022-10-16:
So it works on stock? hmmm...

Orion81, 2022-10-16:
> So it works on stock? hmmm...

Yes. I don't Know why. Any suggest?

Orion81, 2022-10-17:
Yesterday I made a test also with Jelos and it finds the server pc and it pairs, but doesn't download the list of games.
With anbernic works fine.
Jelos and Anbernic doesn't ask the name or ip of the server pc.
Ark Os no way to pair. Before I had e different name of pc (dektop-xxx). Now I change it  in to a little bit short and easy (casa) and I doesn's work anyway. How can I know the IP adress? I want to try with it. The server pc and router are not connected to internet (when I need it I use tethering).
(sorry for my very terrible english)

christianhaitian, 2022-10-17:
Assuming you're using a Windows PC, see here: https://support.microsoft.com/en-us/windows/find-your-ip-address-in-windows-f21a9bbc-c582-55cd-35e0-73431160a1b9#Category=Windows_10

Orion81, 2022-10-17:
[image]

christianhaitian, 2022-10-17:
yea something is not right somewhere as I don't have any of these struggles on my end.

Orion81, 2022-10-17:
Maybe moonlight works only with 5ghz connection?
My router is 2.4 GHz, but I don't think is this the problem (my chrome book connected with no problems). I usually use moonlight trought powerline Lan cabel and a steamlink with no problems.
So the issue is rg503+wi fi. Probably a ip conflict because ge force experience doesn't work without internet connection. Probably my smartphone connected by USB tethering create same ip conflict

Orion81, 2022-10-20:
There was a ip conflict.  With another router works

christianhaitian, 2022-10-22:
Great.  Enjoy!

closed 2022-10-22 by christianhaitian