christianhaitian/arkos #567: Bluetooth not working on RG353V

sksonic, 2022-10-21:
The Bluetooth Manager opens but when go to "Connect to a new Bluetooth device", it waits for a few seconds showing "scanning for available bluetooth devices" but nothing happens and it returns to the menu.

I have tried the same device with Android (stock) and the Anbernic Linux and was able to successfully pair with my controllers.

On the device, I see bluetoothctl does not register any devices. Also attached dmesg logs
[dmesg.txt](https://github.com/christianhaitian/arkos/files/9835843/dmesg.txt)

```
ark@rg353v:/opt/system$ bluetoothctl
Agent registered
[bluetooth]# show
No default controller available
[bluetooth]# exit
ark@rg353v:/opt/system$ sudo hciconfig -a
ark@rg353v:/opt/system$ rfkill list
0: bt_default: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
```

christianhaitian, 2022-10-21:
Enable Bluetooth first by doing F+R3.  You can confirm bluetooth is on by pressing start and verifying it says BT:On

sksonic, 2022-10-21:
Works, thanks! Might be worth updating the FAQ on this.

closed 2022-10-21 by sksonic

christianhaitian, 2022-10-21:
Good point.  It’s been a common question.