christianhaitian/arkos #569: Dos-Pure issue

Industrialchild, 2022-10-22:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?  Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?  Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?  Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.  
Hello and thank you for your time, in advance.  The issue is with the DOS Pure emulator.  There were no issues with it until I tried loading a game called "Seasons of the Sakura" up with it.  It loaded, but with extreme latency in both audio and video.  I quickly got out of the game and deleted the file.  However, to my dismay and for whatever reason, this Audio/Video latency persisted into other MS-DOS games that ran fine before.  This left my modest list of DOS roms unplayable.  The only way I know how to fix this is to completely back-up the micro-SD cards (Samsung Evo brand) that I'm using, format and "clean" the cards using DiskPart on Windows and flashing the latest Ark-OS Disk Image to the cards and then copy my back-ups into the appropriate folders.  This would be very time consuming and I wanted to reach out to you to see if there was a simpler and less time consuming way to fix this issue with this particular emulator.  Thank you for your time!    

### How can the issue be reproduced? (If applicable)
By obtaining a DOS rom of Seasons of the Sakura and running that in DOS-Pure

### What device are you using?
I am using a rg351MP

### What version of ArkOS are you on?  The latest version and I believe it's Version 2.0
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?  Sure, as you know, this particular Ambernic model handheld can utilize 2 mirco-SD cards.  Once this issue with DOS-Pure is triggered, it will even affect that playability of any MS-DOS roms that you have installed on the other card as well.  I've tried deleting the roms of the card(s) and fresh installing them, to no avail.  This issue did happen once with another game (I forget the name of the other rom that caused this issue in the past) and after trying different things, I was only able to fix it by the method described above.

christianhaitian, 2022-10-22:
This sounds odd.  Are you using an authentic good brand sd card such as Samsung or Sandisk?  If so, have you tried updating the Dosbox-pure core via retroarch yet?

Industrialchild, 2022-10-22:
Hello Christian and thanks for taking the time to reach out to me.  To
answer your question, yes, both sd cards in my Ambernic rg351MP are Samsung
Evo 256 GB mirco-sd cards.  I know for a fact these are not knock offs
either.  They are great mirco-sd cards, no doubt.  I also at first did try
to solve this issue by updating the core via Retroarch on the handheld
directly.  That did not solve the problem.  The second thing I tried was to
remove the core, re-install the core and for good measure update it again
via Retroarch's interface.  Both of these solutions did not fix the issue.
Ultimately, to date, the only way I know how to fix this issue is to use
DiskPart on Windows to completely erase all data on both cards, use Rufus
to flash the latest ArkOS disk image and copy my backups of my roms into
the appropriate folders.  I know this sounds odd and yes, it is quite odd
to say the least.  I didn't expect this to happen a second time with the
Seasons of the Sakura rom, but it did.  I believe trying to run that
particular rom with the DOSbox-Pure emulator is a sure fire way to trigger
this strange bug.  Thankfully, this bug, when it happens, only seems to
have this negative effect while running other games with DOSbox-Pure.  I
don't see any other emulators suffer performance wise.

I forgot to mention this also, but even updating the firmware or
downgrading the firmware, via the use of a .tar ArkOS file and the updates
folder, does not seem to remedy this situation.  I'm not the most "tech -
savvy" guy, but this would lead me to believe that the issue would have to
arise from some sort of file generated somewhere either in one or a few of
the many folders that make up the system folders related to databases and
kernels that allow DOSbox-Pure to function as is intended for the
particular hardware said firmware is intended for use on.

If this is the case, I would not know where to begin to look to solve this
issue.

I hope all of this information helps you Christian and it's not a problem
at all if you cannot figure out a solution.  I still absolutely love your
firmware and have been happily using it for around a year and a half on my
handheld!  Kudos also on implementing a "swapfile" script, allowing me to
store and access games for a plethora of systems of both mirco-sd cards so
effortlessly!  It's one of many amazing features of ArkOS and I could not
be happier with exclusively using it on my rg351MP!

Thanks again for your valuable time and attention regarding this matter.

SIncerely,
David Martin

christianhaitian, 2022-10-22:
Thanks for the information.  I just built an updated version of the Dosbox pure core and uploaded it to the repo.  Can you update the cores and try again and see if the issue still occurs?

christianhaitian, 2022-10-30:
Greetings, any update on this?

closed 2022-11-19 by christianhaitian