christianhaitian/arkos #571: Portmaster - Sonic Mania shows black screen

waeggy123, 2022-10-25:
Bug:
Sonic Mania shows black screen in Ports 
(Installed via Portmaster, Data.rsdk added)
see log file
[image]

Device:
RG351V
ArkOS 2.0 09052022
Portmaster 6.58

waeggy123, 2022-10-25:
[image]

waeggy123, 2022-10-25:
Settings.ini —> pixWidth above 400 (default: 320) shows full screen (black bars on the top and bottom)

christianhaitian, 2022-10-26:
If you go to video and turn full screen on, it fixes the issue but then sets the game into widescreen mode.  My understanding is that during gameplay, the default settings work fine for full screen on the 351v, just the menus are too big.  Not much we can do to fix that since the port is designed to work on widescreen devices, not 4:3.

closed 2022-10-26 by christianhaitian