christianhaitian/arkos #572: Duckstation core does no longer launch on RG351V

halpz, 2022-11-02:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Duckstaion used to work, but since I updated the OS, it no longer boots, it shows the arkos splash screen twice before returning to the playstation menu.

I have the required bios files as it was working before

I've tried deleting and re-downloading the core in retroarch with no avail

### How can the issue be reproduced? (If applicable)

select playstation game with duckstation retroarch as the core
doesnt load

### What device are you using?
RG351V

### What version of ArkOS are you on?
arkos 2.0 (10292022)
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2022-11-02:
I can't replicate this issue on my end on the same unit with the same latest version of ArkOS.  Are you able to launch your games with pcsx-rearmed?  If not, something is probably up with your games.

halpz, 2022-11-02:
they work fine with pcx-rearmed. When I launch retroarch standalone, an try to 'launch core' with duck station, that fails also, so I think it is something to do with the core or the config - are there any other steps I can take to try and 'reset' it ?

halpz, 2022-11-02:
also PCSX_rearmed only appears in retroarch32, even though it looks like it's installed in retroarch

halpz, 2022-11-02:
is there a system log I can read to see what's happening?

christianhaitian, 2022-11-02:
In the options menu, you can try restoring the Retroarch Core Settings to see if that resolves the issue for you.  I suspect you have a core option that is set that is causing that issue but I may be wrong.

PCSX_rearmed is currently configured for just retroarch32 in the emulationstation menu since it was primarily a 32bit emulator that recently got a 64bit build that is performing mostly acceptably now.  I'll add that as an additional selectable core for psx in the emulationstation menu in the future.

No log is kept.  What you can do is ssh into the device and run the following command and see what it returns to you:
`retroarch -v -L /home/ark/.config/retroarch/cores/duckstation_libretro.so <game location/game name>`.

halpz, 2022-11-02:
I found the issue - it looks like the cores no longer look in /roms2/bios/ for bios (this worked before), I moved them to /roms/bios and it's now working fine

thanks for the ssh tip, that helped me!

christianhaitian, 2022-11-02:
Did you change this in the retroarch menu as that's set by default when you switch to sd2?  Either way, glad it's working now.

closed 2022-11-02 by christianhaitian

halpz, 2022-11-03:
@christianhaitian I didn't change anything, it seemed to change after one of the updates, I won't touch anything now as it's working!