christianhaitian/arkos #575: AM2R not loading on OGA 1.1

scheckley, 2022-11-07:
Odroid Go Adance black edition (OGA v1.1)

ArkOS 2.0 11062022

AM2R installed as in the ports wiki instructions. am2r.apk built from am2r 1.1 using am2r launcher is copied to gamedata folder.

Selecting AM2R from the ports menu returns back to the menu.

Is there any way to debug?

Thanks.

scheckley, 2022-11-07:
I've replaced the AM2R.sh with this code based on https://github.com/christianhaitian/arkos/blob/main/ports/docs/packaging.md:

`sudo chmod 666 /dev/tty1

param_device="oga"
directory="roms"
export LD_LIBRARY_PATH=/$directory/ports/am2r/libs:/usr/lib
sudo rm -rf ~/.config/am2r
ln -sfv /$directory/ports/am2r/conf/am2r/ ~/.config/
cd /$directory/ports/am2r
sudo ./oga_controls gmloader $param_device &
SDL_GAMECONTROLLERCONFIG="$sdl_controllerconfig" ./gmloader gamedata/am2r.apk
sudo kill -9 $(pidof oga_controls)
unset LD_LIBRARY_PATH
sudo systemctl restart oga_events &
printf "\033c" > /dev/tty1`

AM2R runs fine on the oga 1.1 with this rather than the default:

`#!/bin/bash
GAMEDIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )/am2r"
LIBDIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )/am2r/lib32"
BINDIR="/opt/box86"

# gl4es
export LIBGL_FB=4

# system
export LD_LIBRARY_PATH=$LIBDIR:/usr/lib32:/usr/local/lib/arm-linux-gnueabihf/

# box86
export BOX86_ALLOWMISSINGLIBS=1
export BOX86_LD_LIBRARY_PATH=$LIBDIR
export BOX86_LIBGL=$LIBDIR/libGL.so.1
export BOX86_PATH=$BINDIR

cd $GAMEDIR

sudo ./oga_controls &
sudo systemctl start skhotkey.service
$BINDIR/box86 ./am2r
sudo kill -9 $(pidof oga_controls)
sudo systemctl stop skhotkey.service
sudo systemctl restart oga_events &`

christianhaitian, 2022-11-07:
How did you install portmaster?

scheckley, 2022-11-08:
It was a while back, but I'm pretty sure it was however the system was configured after installation.  I will do a fresh installation on the odroid and check.

scheckley, 2022-11-14:
sorry, it seems to have been a local issue with port master and a clean install of ArkOS and port master has corrected the problem.

closed 2022-11-14 by scheckley