christianhaitian/arkos #578: ArkOS install images should include /lib/arm-linux-gnueabihf/librga.so.2

mjkaye, 2022-11-16:
ArkOS install images should include `/lib/arm-linux-gnueabihf/librga.so.2`. This can be a symlink to `/lib/arm-linux-gnueabihf/librga.so`.

If you build `retroarch32` using the scripts from https://github.com/christianhaitian/rk3326_core_builds or https://github.com/christianhaitian/rk3566_core_builds then the above mentioned symlink is created (in the chroot) and linked against. This means that a `retroarch32` binary, produced using the default method from either of those repos, will not run on an ArkOS system.

christianhaitian, 2022-11-16:
odd as retroarch32 builds in my chroot using the same scripts run just fine.  Simple enough to add though.

mjkaye, 2022-11-17:
It's curious that you get different results. After a clean retroarch32 build, what output do you get from `ldd retroarch32/retroarch32 | grep librga`?

I run Debian Buster chroots running on a Debian Buster (amd64) desktop, but I get similar results when using your Ubuntu VirtualBox (it links against librga.so.2 but it's a real file rather than a symlink).

In any case, including the previously mentioned symlink in the install images still seems like the best solution.

christianhaitian, 2022-11-17:
> It's curious that you get different results. After a clean retroarch32 build, what output do you get from `ldd retroarch32/retroarch32 | grep librga`?
>
> I run Debian Buster chroots running on a [...]

`ldd /opt/retroarch/bin/retroarch32` = not a dynamic executable

mjkaye, 2022-11-17:
> `ldd /opt/retroarch/bin/retroarch32` = not a dynamic executable

Hmm. That explains the difference in behaviour. However, the default script definitely produces a dynamically linked binary for me. Do you perform an additional step?

christianhaitian, 2022-11-17:
not at all.  Just generate the binary then copy it to the device at /opt/retroarch/bin

mjkaye, 2022-11-17:
I can't work out why we're getting different results. Nevertheless, I still think it would be good to have the symlink in the install images if you're happy to do that.

christianhaitian, 2022-11-19:
This has been accommodated with today's OTA update.

closed 2022-11-19 by christianhaitian

mjkaye, 2022-11-19:
Thanks Christian. Much appreciated. I've tested it successfully on an RG503.

Incidentally, I've realized that the reason you got the result `not a dynamic executable`, when running `ldd /opt/retroarch/bin/retroarch32` on your device, is that `ldd` is a 64-bit application being asked to provide data on a 32-bit binary. Running `ldd` in the 32-bit chroot would work.