christianhaitian/arkos #580: RG353M Wifi not working "Activation failed: Secrets were required, but not provided"

daniellozynski, 2022-11-17:
When trying to connect to Wifi (I tried 2 different networks) I get this error message:

"Activation failed: Secrets were required, but not provided"

The wifi passwords contains no special characters. Wifi is working with the stock OS. Reboot/Re-Install did not help. 

Anyone else having this issue?

daniellozynski, 2022-11-17:
It is a WPA2 and WPA3 activated Wifi. I switched to WPA2 only - and it works. But somehow it does work with WPA2/WPA3 with the stock image still...

christianhaitian, 2022-11-17:
Interesting.  Some others have reported this issue but could never replicate the cause.  I'll see if I can test this somehow but I don't have any wpa3 capable equipment in my possession.

daniellozynski, 2022-11-18:
When I change the WPA-Mode to "WPA2 only" it actually works

sweaterr, 2022-11-23:
How do I Change WPA-Mode to WPA2 only?

I can't see related option.

mBouds, 2022-11-24:
Same, had to disable WPA3 support on my router for it to connect

brunohkbx, 2022-11-26:
See my comment 
https://github.com/christianhaitian/arkos/issues/570#issuecomment-1307347881

luciano2000, 2022-11-28:
same error for me, and I'm using a mesh router with a single signal with 5 and 2.4Ghz.
Stock firmware is ok, but I preffer Ark :) can anyone help ?

christianhaitian, 2022-12-02:
I've updated the wifi configuration script to account for networks with wpa2/wpa3 configuration.  You can update online or simply redownload the ArkOS image for your device to have this updated configuration script.

luciano2000, 2022-12-02:
> I've updated the wifi configuration script to account for networks with wpa2/wpa3 configuration. You can update online or simply redownload the ArkOS image for your device to have this updated configuration script.

Thanks bro, you're fantastic, but I've redownload and the problem persists. 
Is the image actualized ? (The name and date is the same in the download wiki >>> RG353M - [GDrive](https://drive.google.com/file/d/1ZXtNSkuU6J54EZWbUlkiJPzYReFCo059/view?usp=share_link) | [Mega](https://mega.nz/file/HApXiZAS#YHaQhyymQGJaetx4FlMsZ0Uz0-KGMdezc8e0_E73JQM) | [Torrent-N/A] (Updated 11/18/2022) MD5:60E1A984F9309E63E0720C4BB8FFE2BD 

thanks

christianhaitian, 2022-12-02:
It’s correct.  All I did was update the existing image.  Is your wifi  encryption wpa2 and wpa3?  Or is it just wpa3?

BooBerry, 2022-12-02:
Updated, then re-enabled WPA3 on my network and it's connecting fine.

Thanks a lot! :)

truekaerf, 2022-12-03:
I've been having the same issue.  Tried today's build but with no luck.  I'm using WPA2.  Could it possibly be something with having a mesh wifi setup?  I did have it working previously on JelOS.

christianhaitian, 2022-12-03:
with wpa2, it should work without issue.  Not sure on this specific issue with your mesh.

truekaerf, 2022-12-03:
I spent about 4 hours last night trying to debug, went as far as re-enabling journald and turning on logging for NetworkManager.  As best as I can tell, and this is pretty much a guess, it gets confused about what AP it is talking to while trying to establish the connection.  I'm using a TP-Link/Deco home wifi mesh APs.  Tried changing a bunch of settings on my network with no luck.  Every attempt ended in the same "Secrets were required..." message.

Is there anything I can do/provide to maybe help with finding a solution?  I'd really like to move to ArkOS since JelOS is moving to a code only distribution.

luciano2000, 2022-12-05:
Same problem here. I use mesh technology with WPA/WPA2 and seamless 2.4/5ghz band (unique NET, but I've activated compatibility mode too).
Sharing Wifi from my iPhone works with ArkOs
Using stock firmware all works fine.

derekpurdy, 2022-12-23:
Having the same issue with latest download on RG353V

christianhaitian, 2022-12-23:
Try disabling fast roaming if you have that feature turned on in your wireless access point.

derekpurdy, 2022-12-23:
> Try disabling fast roaming if you have that feature turned on in your wireless access point.

That seems to be it! Thanks!

bcyeager, 2023-01-24:
> Try disabling fast roaming if you have that feature turned on in your wireless access point.

This fixed it for me as well on a RG353M.

trr, 2023-02-04:
From my research it turns out that networks with 802.11r fast roaming present another authentication type choice to the client. Clients not supporting that feature should work by just authenticating the normal way eg wpa2 and ignore auth types they don't recognise.

This suggests to me this issue sounds like the same as the wpa3 issue and there may be a similar workaround.

Would love to get wifi working on arkos as on other firmwares but don't know any further than this. My workaround so far has been to add another ssid with 802.11r disabled on my network, or do wifi hotspot from my phone if it's a wifi network I don't admin.

cidus, 2023-09-07:
I also have a mesh wi-fi network with 802.11r (TP-link Deco M5). I'm only able to connect my RG353PS by disabling 802.11r.

If you need someone to test new wi-fi drivers, I volunteer, just hit me up with some pointers and I'll gladly test it out.

mexusbg, 2023-09-19:
Fast roaming work on JELOS and Unofficial OS.
For example this driver works:
https://github.com/morrownr/8821cu-20210916
PKG_VERSION="5a39cef"

If you make a build I can test if it works. Not disabling fast roaming for this firmware.

christianhaitian, 2023-10-02:
For this chipset (8821cs), JelOS uses the same driver used in ArkOS.  8821cu is a different chipset.  If someone doesn't mind, can someone try this?  It will update 2 network components, wpasupplicant and NetworkManager to newer versions.  Perhaps this may help with this issue.  

1.  Just unzip the attached file and place the script to your roms/tools folder (or roms2/tools folder if you're using a 2 sd card setup). 
2.  Go to the Options/Tools section and press A on the NetManUpdate script.  You'll need to make sure you're connected to your wireless network before you try this.  
3.  Once completed, reboot and let me know if you notice any improvement, worsening, or no difference to the issue you reported.

Thank you to anyone who can assist with this.

[NetManUpdate.zip](https://github.com/christianhaitian/arkos/files/12778704/NetManUpdate.zip)

mexusbg, 2023-10-02:
> For this chipset (8821cs), JelOS uses the same driver used in ArkOS. 8821cu is a different chipset. If someone doesn't mind, can someone try this? It will update 2 network components, wpasupplicant [...]

I ran the script while connected to a network without roaming enabled, then tried to connect to a network with roaming but still the same issue. I've read that it's a flag in the driver. Can you try the driver from my previous comment and check the unofficial os mk file for this driver?

cidus, 2023-10-03:
Also tried the script, couldn't connect either with fast roaming.

Any logs that could be helpful? Tried dmesg, this is what comes out:

[  340.730158] RTW: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
[  340.733060] RTW: recv eapol packet 1/4
[  341.742089] RTW: recv eapol packet 1/4
[  341.751761] RTW: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:2, no_gkey_mc_cnt:0
[  342.754845] RTW: recv eapol packet 1/4
[  342.777505] RTW: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
[  343.766303] RTW: recv eapol packet 1/4
[  343.800612] RTW: rtw_aes_decrypt(wlan0) no_gkey_bc_cnt:3, no_gkey_mc_cnt:0
[  344.766659] RTW: OnDisassoc(wlan0) reason=2, ta=68:ff:7b:84:06:a2
[  345.100877] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  345.576312] ========== ACS (VER-3) ==========
[  345.576321] Best 24G Channel:12
[  345.576323] Best 5G Channel:40

[  345.588313] RTW: nolinked power save enter

christianhaitian, 2023-10-03:
Here's a package that updates the wpa_supplicant to version 2.10 similar to as used in JELOS and UnofficialOS.  Try restoring this backup, reboot, and see if it connects to your fast roaming enabled wireless network now.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and test on your wireless network with fast roaming enabled.

Let me know your results.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12788385/arkosbackup.tar.gz)

cidus, 2023-10-03:
That seems to fix it! I'm going to test more, but looking good so far.

cidus, 2023-10-03:
Turned it off and on a few times, transferred 2 gigs of files using SFTP, all good.

christianhaitian, 2023-10-03:
Awesome.  Hopefully someone else can chime in and confirm this is working for them as well.

trr, 2023-10-04:
After running the arkosbackup.tar.gz update, I'm getting no wifi at all, not even showing any ssids.

But, one possible difference is I did that on a clean arkos system - presumably I was supposed to do it after the other update.  I'll re-image and try that

trr, 2023-10-04:
Ok so I started again and applied both fixes in order, and now it does successfully connect to a wifi network that has fast roaming!

It doesn't appear to me that it is actually doing fast roaming, but that's ok.  The fact it'll connect in the presence if fast roaming enabled is good.

christianhaitian, 2023-10-04:
Nice.  Thanks for reporting.  Whether it will roam properly between APs is dependent on the device's wifi compatibility with the wireless network to do it or not.

christianhaitian, 2023-10-07:
Today's update includes this fix.  I'll leave this issue open for a few days in case anyone reports that they're still having issues with connecting to their wifi related to having a mesh wireless system.

christianhaitian, 2023-10-15:
Closing as the issue seems to have been resolved for most with the 10062023 update.

closed 2023-10-15 by christianhaitian