christianhaitian/arkos #583: Certain SD cards fail to be initialized when used as a secondary SD card

oOblik, 2022-11-20:
dmesg output:

[ +36.277793] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[  +0.863627] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[  +0.005888] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[  +0.049847] mmc1: error -110 whilst initialising SD card
[  +0.508237] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[  +0.005914] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[  +0.026402] mmc_host mmc1: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[  +0.527705] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[  +0.005885] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[  +0.022588] mmc_host mmc1: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[  +0.531486] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[  +0.005869] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)
[  +0.026420] mmc_host mmc1: Bus speed (slot 0) = 100000Hz (slot req 100000Hz, actual 100000HZ div = 0)
[  +0.539709] dwmmc_rockchip ff380000.dwmmc: Busy; trying anyway
[  +0.005871] mmc_host mmc1: Timeout sending command (cmd 0x202000 arg 0x0 status 0x0)

### How can the issue be reproduced?

Inserting an SD Card of the following type in the secondary SD card slot:
SanDisk 256GB Ultra microSDXC UHS-I Memory Card - 120MB/s, C10, U1, Full HD, A1, Micro SD Card SDSQUA4-256G-GN6MN 

### What device are you using?
PowKiddy RGB20S

### What version of ArkOS are you on?
ArkOS 2.0 (11182022)

### Anything else you'd like to include that may help to review this issue or feature request?
I've tried using the same SD card as an OS drive, flashed with the same version of ArkOS and it works fine as the primary boot device.

christianhaitian, 2022-11-20:
I don't support the rgb20s.  I can share that this unit is known to have issues with high quality sd cards.  It's best to use lower quality cards on that unit and the second sd seems to work best with fat32.  Good luck.

closed 2022-11-20 by christianhaitian

RAS-GIT, 2025-09-07:
Hello, it seems I fixed this for the R36S clone (panel 5).
The issue was with the voltage regulation on ff370000.
I removed these lines in dwmmc@ff370000 in the dtb file, and now the device loads without any problems with two SD cards:
`sd-uhs-sdr12; sd-uhs-sdr25; sd-uhs-sdr50; sd-uhs-sdr104;`

Here is my dwmmc@ff370000 section:
```
	dwmmc@ff370000 {
		compatible = "rockchip,px30-dw-mshc", "rockchip,rk3288-dw-mshc";
		reg = <0x00 0xff370000 0x00 0x4000>;
		max-frequency = <0x8f0d180>;
		clocks = <0x02 0xf7 0x02 0x3b 0x02 0x43 0x02 0x44>;
		clock-names = "biu", "ciu", "ciu-drv", "ciu-sample";
		assigned-clocks = <0x02 0x3b>;
		assigned-clock-parents = <0x02 0x57>;
		power-domains = <0x87 0x07>;
		fifo-depth = <0x100>;
		interrupts = <0x00 0x36 0x04>;
		pinctrl-names = "default";
		pinctrl-0 = <0x8c 0x8d 0x8e 0x8f>;
		status = "okay";
		bus-width = <0x04>;
		cap-mmc-highspeed;
		cap-sd-highspeed;
		supports-sd;
		card-detect-delay = <0x320>;
		ignore-pm-notify;
		cd-gpios = <0x59 0x03 0x01>;
		vqmmc-supply = <0x49>;
		vmmc-supply = <0x90>;
		phandle = <0x105>;
	};
```

Original section:
```
	dwmmc@ff370000 {
		compatible = "rockchip,px30-dw-mshc", "rockchip,rk3288-dw-mshc";
		reg = <0x00 0xff370000 0x00 0x4000>;
		max-frequency = <0x8f0d180>;
		clocks = <0x02 0xf7 0x02 0x3b 0x02 0x43 0x02 0x44>;
		clock-names = "biu", "ciu", "ciu-drv", "ciu-sample";
		assigned-clocks = <0x02 0x3b>;
		assigned-clock-parents = <0x02 0x57>;
		power-domains = <0x87 0x07>;
		fifo-depth = <0x100>;
		interrupts = <0x00 0x36 0x04>;
		pinctrl-names = "default";
		pinctrl-0 = <0x8c 0x8d 0x8e 0x8f>;
		status = "okay";
		bus-width = <0x04>;
		cap-mmc-highspeed;
		cap-sd-highspeed;
		supports-sd;
		card-detect-delay = <0x320>;
		ignore-pm-notify;
		cd-gpios = <0x59 0x03 0x01>;
		sd-uhs-sdr12;
		sd-uhs-sdr25;
		sd-uhs-sdr50;
		sd-uhs-sdr104;
		vqmmc-supply = <0x49>;
		vmmc-supply = <0x90>;
		phandle = <0x105>;
	};
```

Honestly, I don't know how this will affect stability; probably more thorough testing is needed.