christianhaitian/arkos #584: how do I connect to wifi?

brothan323, 2022-11-20:
I go into options -> wifi -> set the exact password and for some reason I keep getting a "activations failed: Secrets are required" error.

????????

christianhaitian, 2022-11-20:
If you have wpa3 enabled in your wireless environment, it will cause issues. Try making sure wireless security settings are set to wpa2

brothan323, 2022-11-20:
I got it. Working. Another questions I have is that is it possible to update retroarch? It's missing a bunch of cores and using a very old version.

christianhaitian, 2022-11-20:
very old version of retroarch? the retroarch on here is the latest 1.12 version and many of the cores are only a few months old.

brothan323, 2022-11-20:
O I say very old version as it's using an old UI. It's fine though. One last question, how do you adjust the sensitivity of the joysticks on the standalone emulators?

christianhaitian, 2022-11-20:
That will depend from standalone emulator to standalone emulator.  Some may not even allow that change.  ppsspp allows that change from the emulator menu accessible by hitting the r3 button while in game.
you can change the ui for retroarch via the retroarch settings>driver>menu.  you can set it to xmb or others.  Also, you can update to the latest version of retroarch cores that are available from our central repo  from the retroarch menu as well.  Make sure to check the emulators and ports section of the wiki for more info on available emulators.

good luck and have fun.

closed 2022-11-20 by christianhaitian