christianhaitian/arkos #585: 353m Emulator Settings (emu and core selection) not working

SynergyGamingTV, 2022-11-20:
ArkOS 2.0 (11182022) -- 353m

When trying to change the emulator settings to select the default emulator and core through emulation station (start > emulator settings), the settings don't seem to apply to the selected rom list.  Manual individual emulator and core selection works through the "edit metadata".  I am having the issue with NES and Neogeo.  Other systems not tested.

christianhaitian, 2022-11-20:
Changing the settings works just fine on my end with the 353m.  You sure you haven't set specific emulators for each of your roms which will take precedence?

SynergyGamingTV, 2022-11-22:
Hi there, 

Thank you for the email.  I’m finding that it’s causing the individual ROMs to have no “emulator” selected when looked at the metadata of each game.  However, when I set the individual game to the “default” it is the same as the overall setting in the “emulator settings”, as in they do match.

Any ideas as to why some games wouldn’t have an emulator selected?

From: christianhaitian
Sent: November 20, 2022 4:20 PM
To: christianhaitian/arkos
Cc: SynergyGamingTV; Author
Subject: Re: [christianhaitian/arkos] 353m Emulator Settings (emu and coreselection) not working (Issue #585)

Changing the settings works just fine on my end with the 353m. You sure you haven't set specific emulators for each of your roms which will take precedence?

christianhaitian, 2022-11-22:
Did you transfer your games from another distribution that also used EmulationStation?  If so, that’s probably your issue.  To resolve, you’d have to strip those specific emulator settings out or simply delete the gameslist file in each rom folder.  Be aware in deleting the gamelist, you’ll lose the link to any scrapped images.

SynergyGamingTV, 2022-11-22:
Right, so why would my roms not have an emulator selected.  They literally crash to menu unless I manually set each one.  The defaults don’t override the “Blank” emulator in the metadata?

From: christianhaitian
Sent: November 22, 2022 8:43 AM
To: christianhaitian/arkos
Cc: SynergyGamingTV; Author
Subject: Re: [christianhaitian/arkos] 353m Emulator Settings (emu and coreselection) not working (Issue #585)

Also, no specific emulator set just means it will use the system default in ES.

christianhaitian, 2022-11-22:
They should.  I suspect something may be wrong with your gamelist.xml file.  Have you tried deleting it or renaming it to something else to see if they work after ES recreates the gamelist.xml file?

SynergyGamingTV, 2022-11-22:
That was it, gamelist.xml file.  Yes they were transferred over from my 351.  Thank you so much for taking the time to assist.  I consider myself pretty good at figuring stuff out, but that was a little beyond my knowledge.  

Would you like me to post anything on that tread oh GitHub, or should it just be removed?

Thanks again!

From: christianhaitian
Sent: November 22, 2022 8:42 AM
To: christianhaitian/arkos
Cc: SynergyGamingTV; Author
Subject: Re: [christianhaitian/arkos] 353m Emulator Settings (emu and coreselection) not working (Issue #585)

Did you transfer your games from another distribution that also used EmulationStation? If so, that’s probably your issue. To resolve, you’d have to strip those specific emulator settings out or simply delete the gameslist file in each rom folder. Be aware in deleting the gamelist, you’ll lose the link to any scrapped images.

christianhaitian, 2022-11-23:
Great.  I’ll close this issue.  Enjoy!

closed 2022-11-23 by christianhaitian