christianhaitian/arkos #587: Bluetooth audio disable buttons

ftoledoroa, 2022-11-23:
Since the 11.18 update, RG353VS recognizes Bluetooth headphones as a controller (I think) and when starting retroarch the console buttons stop working. In Emulation station still working.  
The buttons work when I not using Bluetooth audio.
I have tried removing other Bluetooth controls but the problem persists.

christianhaitian, 2022-11-23:
Oh wow, you’re right. I’ll get a resolve for this out asap.  Thanks for reporting .

JuanMiguelBG, 2022-11-23:
My fix:

"/usr/local/bin/retroarch" file

`#!/usr/bin/env bash

/usr/local/bin/controller_setup.sh

EXTRA_CONFIG=""

if [[ $(amixer scontrols | grep -c " A2DP'") -gt 0 ]]; then
  if [ "$(ls /dev/input/ | wc -l)" -gt "9" ]; then
    EXTRA_CONFIG="--appendconfig=/home/ark/.config/retroarch/retroarch.cfg.ext"
  fi
elif [ "$(ls /dev/input/ | wc -l)" -gt "7" ]; then
  EXTRA_CONFIG="--appendconfig=/home/ark/.config/retroarch/retroarch.cfg.ext"
fi

/opt/retroarch/bin/retroarch -c /home/ark/.config/retroarch/retroarch.cfg $EXTRA_CONFIG "$@"
`
"/usr/local/bin/retroarch32" file

`#!/usr/bin/env bash

/usr/local/bin/controller_setup.sh

EXTRA_CONFIG=""

if [[ $(amixer scontrols | grep -c " A2DP'") -gt 0 ]]; then
  if [ "$(ls /dev/input/ | wc -l)" -gt "9" ]; then
    EXTRA_CONFIG="--appendconfig=/home/ark/.config/retroarch32/retroarch.cfg.ext"
  fi
elif [ "$(ls /dev/input/ | wc -l)" -gt "7" ]; then
  EXTRA_CONFIG="--appendconfig=/home/ark/.config/retroarch32/retroarch.cfg.ext"
fi

/opt/retroarch/bin/retroarch32 -c /home/ark/.config/retroarch32/retroarch.cfg $EXTRA_CONFIG "$@"
`

"/usr/local/bin/retroarchtate" file

`
#!/bin/bash

if [[ "$@" == *"fbneo_libretro"* ]]; then
  vertmode=$(cat /home/ark/.config/retroarch/retroarch-core-options.cfg | grep "fbneo-vertical-mode" | grep -oP '"\K[^"]+')
  sed -i '/fbneo-vertical-mode \=/c\fbneo-vertical-mode \= "alternate"' /home/ark/.config/retroarch/retroarch-core-options.cfg
  core="fbneo"
elif [[ "$@" == *"mame2003_plus_libretro"* ]]; then
  vertmode=$(cat /home/ark/.config/retroarch/retroarch-core-options.cfg | grep "mame2003-plus_tate_mode" | grep -oP '"\K[^"]+')
  sed -i '/mame2003-plus_tate_mode \=/c\mame2003-plus_tate_mode \= "enabled"' /home/ark/.config/retroarch/retroarch-core-options.cfg
  core="mame2003_plus"
fi

/usr/local/bin/controller_setup.sh

EXTRA_CONFIG=""

if [[ $(amixer scontrols | grep -c " A2DP'") -gt 0 ]]; then
  if [ "$(ls /dev/input/ | wc -l)" -gt "9" ]; then
    EXTRA_CONFIG="--appendconfig=/home/ark/.config/retroarch/retroarch.cfg.ext"
  fi
elif [ "$(ls /dev/input/ | wc -l)" -gt "7" ]; then
  EXTRA_CONFIG="--appendconfig=/home/ark/.config/retroarch/retroarch.cfg.ext"
else
  EXTRA_CONFIG="--appendconfig=/home/ark/.config/retroarch/retroarch.cfg.vert"
fi

/opt/retroarch/bin/retroarch -c /home/ark/.config/retroarch/retroarch.cfg $EXTRA_CONFIG "$@"

if [[ "$core" == *"fbneo"* ]]; then
  sed -i "/fbneo-vertical-mode \=/c\fbneo-vertical-mode \= \"${vertmode}\"" /home/ark/.config/retroarch/retroarch-core-options.cfg
elif [[ "$core" == *"mame2003_plus"* ]]; then
  sed -i "/mame2003-plus_tate_mode \=/c\mame2003-plus_tate_mode \= \"${vertmode}\"" /home/ark/.config/retroarch/retroarch-core-options.cfg
fi
`

christianhaitian, 2022-11-23:
Thank you Juan.  I think I found a solution that will test for actual gamepads/joysticks connected.  See my solution below that still needs testing:

/usr/local/bin/retroarch
```
#!/usr/bin/env bash
/usr/local/bin/controller_setup.sh

if [ "$(ls /dev/input/ | wc -l)" -gt "7" ]; then
  testJoy=$(timeout 1 sdljoytest | grep Found)
  testJoy=$(echo "$testJoy" | cut -d " " -f6)
  if [ "$testJoy" -gt "1" ]; then
    /opt/retroarch/bin/retroarch -c /home/ark/.config/retroarch/retroarch.cfg --appendconfig=/home/ark/.config/retroarch/retroarch.cfg.ext "$@"
  else
    /opt/retroarch/bin/retroarch -c /home/ark/.config/retroarch/retroarch.cfg "$@"
  fi  
else
  /opt/retroarch/bin/retroarch -c /home/ark/.config/retroarch/retroarch.cfg "$@"
fi
```

christianhaitian, 2022-11-24:
Here's an update that I hope should fix the controls not working when bluetooth audio is connected.  I haven't had a chance to fully test this as I don't currently have a bluetooth controller available to verify it functions as it should to make sure bluetooth controllers will still work but it will fix the BT audio and missing controls issue. 

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test controls with bluetooth audio enabled.

Let me know how it works out for you.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/10081281/arkosbackup.tar.gz)

JuanMiguelBG, 2022-11-24:
I've tested the changes and works successfully, great work.

jknut, 2022-11-26:
Should this fix also work on the RG353?  I tried it and the gamepad works but I don't have sound out of the Bluetooth headphones (Galaxy Buds2 Pro).

christianhaitian, 2022-11-26:
Yes, it should work for the rg503, rg353m and rg353v.  This should not impact bt audio not working.

ftoledoroa, 2022-11-28:
Hi, I have tried the two solutions proposed with arkosbackup.tar.gz but the Bluetooth adapter no longer recognizes the wireless headphones (Sony LinkBuds S) with both.
Before the 11.18 update it properly recognized headphones.
After the 11.18 update, it recognizes headphones but the buttons do not work.
After doing the recovery with arkosbackup.tar.gz it no longer recognizes the headphones.

christianhaitian, 2022-11-29:
> Hi, I have tried the two solutions proposed with arkosbackup.tar.gz but the Bluetooth adapter no longer recognizes the wireless headphones (Sony LinkBuds S) with both. Before the 11.18 update it [...]

What is your unit?  I can't replicate the issue with headphones not working on my rg503 or rg353m

ftoledoroa, 2022-11-29:
It's a rg353VS

christianhaitian, 2022-11-29:
Did you make sure bluetooth was still on?

ftoledoroa, 2022-11-29:
Of course brother, there are even other devices like an 8bitdo controller.

christianhaitian, 2022-11-29:
So just your bluetooth headset stopped connecting?  Not the 8bitdo controller?

ftoledoroa, 2022-11-29:
exactly, only headphones are not shown.

christianhaitian, 2022-11-29:
And you're sure you've put your headphones in pairing mode when trying to connect?

christianhaitian, 2022-11-29:
The changes made by the backup file above changes nothing with bluetooth connection or disconnection or the bluetooth stack at all.  Just changes with how retroarch reacts when an active bluetooth connection exist so I can not replicate your issue.

ftoledoroa, 2022-11-29:
I will try to flash Ark again and then update to the latest version to apply the solution you gave, so I wanted to know what recommendation you give me to do the process.  Greetings.

ftoledoroa, 2022-11-30:
Hello, I have finally solved the problem.
I have installed ArkOS again on my TF1 card and then updated to the latest version, then I synchronized the Bluetooth headphones (Sony LinkBuds S) and it recognized them properly, obviously in this step the system had the bug of not working the buttons when having the connected headphones, but after applying the solution in the backup arkosbackup.tar.gz it has been solved properly.
So this bug is fixed for the next version, in my humble opinion.
Thanks for the help.
Sorry for the bad English, I speak Spanish.

christianhaitian, 2022-11-30:
Excellent.  I will push an update out with this fix as soon as possible.

christianhaitian, 2022-12-02:
This fix has been pushed in this evening's update.

closed 2022-12-02 by christianhaitian