christianhaitian/arkos #589: Request: add 100Hz panel screen timings for RG353 devices

colpoz, 2022-11-26:
Hi

as already done for the 60Hz, I ask if it is possible to add 100Hz timings options for RG353x
It would be very very important to play European PAL games, which run at 50fps.
Crrently, the only console I can play PAL games on is the RG351V with the original timings at 149hz, which is a near perfect multiple of 50.

The RG353x panel supports 100Hz and someone has already tried to make it work with success.
https://www.mail-archive.com/dri-devel@lists.freedesktop.org/msg413878.html

Thank you!
Stefano

christianhaitian, 2022-11-26:
At this time, you can switch back to the original timing of 120hz.  I don’t know if the dev who provided the 60hz fix will provide a solution for 100hz.  If he does, I will implement it.

colpoz, 2022-11-28:
How can I contact that dev? I would like to try to ask him if he can develop 100hz timings too. Thanks

christianhaitian, 2022-11-28:
It is TonyJih who is also responsible for the 60hz screen timing for the 351mp and 351v units as well.  I don't know how best to contact him other than via the Retro Game Handhelds Discord but he is aware of this request.  He will review and decide if and when he provides a solution for it.  I'm going to close this ticket since it's not a feature I could provide and if it's ever available, I will add it in a future update.  Take care!

closed 2022-11-28 by christianhaitian