christianhaitian/arkos #591: RG353M Tools option missing after switching to 2nd SD for ROMS

kojimon23, 2022-11-30:
"Tools" option disappear once "Switching to 2nd SD for ROMS" option was invoked. Switching back to Main for Roms does not make the "tool" option come back.

RG353M running ArkOS 2.0 (11182022)

christianhaitian, 2022-12-01:
does your 2nd sd have a tools folder?  If so, does it have thememaster and portmaster in it?  If not, you can add them from [here](https://johnirvine1433.github.io/ThemeMaster/) and [here](https://github.com/christianhaitian/PortMaster).  That will then let the tools option folder show in the menu again

kojimon23, 2022-12-01:
Hi,

I just checked and there is a tools folder but it is empty. I will check
out the links and let you know.

Thank you again for your support!

- Ben
ᐧ

closed 2022-12-01 by christianhaitian