christianhaitian/arkos #592: RG353V, Kodi starts with wrong orientation, without input setup when starting from ports area, OK when starting with START - Menu  Kodi Media Center

gautschh, 2022-12-02:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2022-12-02:
Kodi for the rg353v is only available through the start menu,  I don’t know where or how you got kodi through ports.

closed 2022-12-02 by christianhaitian

gautschh, 2022-12-02:
It was my fault, I copied all Files from 351V SD-Card to my new 353V, so Kodi transfered from old sdcard, everything fine, thx for this great alternative OS !