christianhaitian/arkos #594: RG353V: "fsckd-cancel-msg:Press  Ctrl+C  to cancel all filesystem checks in progress"

MetroidFusion, 2022-12-03:
Hi

After updating to the newest build (12012022) for the RG353V, I now get the following message when booting the device: "fsckd-cancel-msg:Press  Ctrl+C  to cancel all filesystem checks in progress"

The device still continues to boot ArkOS, but this message seems to slow down the boot process, and I have not seen it on any device running ArkOS before. As mentioned, this happened immediately after updating to the newest build on the RG353V. 

/M

MetroidFusion, 2022-12-03:
After reading the changelog for this build, I suppose the boot message is related to the following:

"- Adjusted file system check interval for ext4 (root) partition (RG353M/V/VS and RG503 Only!)"

However, this file system check happens every single time the device boots now. Is this intentional? :)

christianhaitian, 2022-12-03:
That filesystem check is for the easyroms partition of tf1.  Some were reporting issues with booting their units after sometime and it seemed to be due to corruption.  After some review, it seemed like it may have been related to the psuedo overclock that doesn't work that I disabled in the same update. I'll push another update out to remove the file system check for the roms partition on tf1.  No need to keep it there if there's no need.

christianhaitian, 2022-12-03:
A new update has been released with this change

closed 2022-12-03 by christianhaitian