christianhaitian/arkos #598: PCSX_Rearmed core bugs

v99B, 2022-12-08:
I think your work is really great, what you do.

I only have one problem that I noticed with the psx core (PCSX_Rearmed) in the two Harry Potter games, none of them work. it only goes as far as the book and no further. Bios is available. It would be really great if they could take care of the problem. I've tried jelos, it works with this core.
so there must be some kind of error.

v99B, 2022-12-08:
I use the RG503

christianhaitian, 2022-12-08:
You can update the core separately through retroarch32.  Just launch retroarch32, then go to Online Updater, then Update Installed Cores.  See if the issue is resolved then.

v99B, 2022-12-09:
worked great thanks for the help.

closed 2022-12-09 by v99B