christianhaitian/arkos #599: RG353V retroarch menu button key not working properly

andikahakim17, 2022-12-10:
Hi,
everytime I change my rg353v hotkey retroarch menu button, it works for the first time, but everytime I quit retroarch and back again, the hotkey button is not bring me to the menu instead quitting the retroarch. This is not happen on rg353m

Regards,

christianhaitian, 2022-12-11:
I need more information.  What version of ArkOS are you running?  What key are you trying to change the retroarch hotkey to?

andikahakim17, 2022-12-11:
Hi,
I’m using the latest version on this github.
I try to bind it to select + start.
and once it got problem, I try to bind it to L1+R1, but still same.
Like I said, the first time it’s working, once I restart the device the hotkey going for quit retroarch

christianhaitian, 2022-12-11:
Are you using the ozone menu driver?  If so, have you done an OTA update yet? The version of ArkOS you should be on now is v2.0 (12102022)

andikahakim17, 2022-12-12:
Yes I'm using ozone, I just done the OTA update and now it fixed. Thanks.

closed 2022-12-12 by andikahakim17