christianhaitian/arkos #601: Some Requests

CeterisParibus51, 2022-12-19:
not sure how feasible these would be.
after i get an adapter from usb c to usb so i can use a keyboard i plan on attempting these manually with terminal if anything works well ill be sure to pass on the info.

including/porting these could expand capabilities by a large amount, not sure how well they would work on the rockchip or any other architectures your working with

wine / PlayOnLinux 
Virtual Box
box86 (its used in some port master stuff already, so adding it natively could reduce redundancy)
fna (open source xna clone)
Anbox (android environment)

including synaptic package manager in tools would possibly really simplify installing some stuff if it will work with the architecture, 

an internet browser would be an amazing addition to tools

and lastly if notepad++ could be added (even through portmaster) that would be fantastic.

a simple way to boot into a desktop environment like gnome could also be really useful but may be excessive and a nightmare to get working properly.

christianhaitian, 2022-12-19:
Hi,

A graphical internet browser would require x11 and I don't have any plans to add x11 support to ArkOS for any supported device in the near future.  Many of the items you're requesting here are not central to the primary goal of the distro which is retrogaming emulation.  box86 is only used for ports at this time so it being installed through portmaster installed games is suffice when needed.  If there is a retrogaming reason to add box86/box64 natively in the future, it will be considered at that time.  I don't think wine/playonlinux will provide any value on any of the currently supported device due to their overhead needs but after your testing, if you find something compelling in relation to retrograming emulation, please feel free to share.  Be aware that if you're going to be building packages directly on the device, you'll want to enable developer mode.  You can do this by either unhiding the Options/Advanced section in Emulationstation and selecting Enable Developer Mode, or following the instructions [here](https://github.com/christianhaitian/arkos/wiki/Building-packages-and-modules-on-your-device).  Enjoy!

closed 2022-12-19 by christianhaitian

CeterisParibus51, 2022-12-19:
thank you for the timely reply and for the tip about developer mode, ill keep you posted if i find anything useful in the context of retro gaming pertaining to wine or the other mentioned packages.