christianhaitian/arkos #603: No hotkey for PPSSPP on RG353v?

danbanasbananabandanas, 2022-12-27:
Issue: Hotkey for PPSSPP doesn't work/is mapped to save state 
FAQ says PPSSPP hotkey is R3, but that just saves state for me (L3 loads state)
I have only just installed this, so it had to be a default setting.
R2 and L2 don't do anything at all.
I thought maybe I was on an older version and updated arkos to 2.0 (12252022), but it still just saves state instead of pulling up the PPSSPP menu. 
I can't seem to configure the PPSSPP hotkey manually either.
I have tried every button and searched everywhere in the wiki to no avail for this. 
I have found someone else that had the same issue on reddit [here](https://www.reddit.com/r/RG353V/comments/zcd2bz/psp_settings_on_arkos/), but no answers

christianhaitian, 2022-12-27:
Your psp content must have been from a previous ArkOS install?  If so, just delete the ppsspp folder within the /roms/psp folder.  If you have game saves you'd like to maintain, just delete the ppsspp/PSP/SYSTEM/controls.ini and ppsspp/PSP/SYSTEM/ppsspp.ini.sdl file and on the next launch, of the ppsspp standalone emulator, those files will be recreated that fix that issue

closed 2022-12-27 by christianhaitian

danbanasbananabandanas, 2022-12-30:
Yes, that was the exact issue and this fixed it! Thanks so much for everything!

guzuv, 2024-06-26:
how about standalone-2021?

CoolHandLukesic, 2024-08-16:
> Your psp content must have been from a previous ArkOS install? If so, just delete the ppsspp folder within the /roms/psp folder. If you have game saves you'd like to maintain, just delete the [...]

I just did these steps. I only deleted the last two to be able to keep my saved data and I have no change.

CoolHandLukesic, 2024-08-16:
I did it once more and it worked!