christianhaitian/arkos #605: Atari 800 configuration RG353M

sprudel79, 2023-01-03:
Hi everyone,
thanks for all the work done in this project!
I have recently switched from RG351MP to RG353M and I have a problem with the configuration provided for the Atari 800 emulator (basically with the controls).
I am using latest updated version of ArkOS (12302022).
With the setup provided for RG351MP I was able to use both dpad and left analog as the joystick, also the common combination select+x to enter the Retroarch menu was possible.
With the RG353M provided config setup only left analog is working as joystick, the onscreen keyboard is reached via dpad down and what I've just figured out is that R3+X is now the combination for the Retroarch menu.
When I reset the controller configuration in the Retroarch settings it seems to be fine again. However, when I save the config my changes are reverted with the next start of any Atari 800 game.
In general it looks like that the Atari 800 emulator is the only one that comes with a predefined override of the controller configuration and perhaps something is preventing me from changing this manually.
Could anybody please assist?
Thanks in advance :-)

christianhaitian, 2023-01-03:
I have a fix coming for this soon.

sprudel79, 2023-01-03:
Hi @christianhaitian, many thanks, sounds great!
I've just donated some money for your awesome work ;-)

christianhaitian, 2023-01-03:
This should be addressed with today's update. Thanks for your donation.

closed 2023-01-03 by christianhaitian