christianhaitian/arkos #607: RG353M won't boot

dmzkrsk, 2023-01-09:
Hello. I downloaded the latest ArkOS image, but my RG353M unit doesn't start at all. Just black screen

- I used different SD cards
- Used both Apple Pi and balena for flashing
- Default SD boots
- JELOS boots fine (from the first attempt)

I found some related threads for other devices:

- https://github.com/christianhaitian/arkos/issues/515
- https://github.com/christianhaitian/arkos/issues/439

dmzkrsk, 2023-01-09:
Also tried `dd` under Linux. I was able to mount `ext4` partition, so I think the flashing was fine. Any way to enable some kind of debug log of boot process?

Partition table looks like

```
/dev/sdc1     16384    24575     8192    4M unknown
/dev/sdc2     24576    32767     8192    4M unknown
/dev/sdc3     32768   237567   204800  100M unknown
/dev/sdc4    237568 14983167 14745600    7G Microsoft basic data
/dev/sdc5  14983168 15521791   538624  263M Microsoft basic data
```

I was not able to mount three first partition though

christianhaitian, 2023-01-09:
Do you get a logo screen at all?  Does it at least do the initial setup?

dmzkrsk, 2023-01-10:
- somehow, pressing power button for some long/short time I was able to boot into logo, but it hang
- I restarted the device and it continued to boot normally
- the next day device died (?) no led, doesn't react to power button, so I send it back

Perhaps, faulty device

closed 2023-01-10 by dmzkrsk

christianhaitian, 2023-01-10:
Did you make sure to fully charge the device first?