christianhaitian/arkos #608: Virtual network Zerotier started incorrectly

jokerwho, 2023-01-12:
In fact, I wanted to stream a Moonlight game remotely over a virtual network and my own public IP address, but after installing Zerotier over ssh and starting it up, I got the following error:

`ERROR: unable to configure virtual network port: could not open TUN/TAP device: No such file or directory `

In the general search solution thick, I found that seems to be the need to compile kernel in terms of steps?

`
make menuconfig
Network core driver support->Universal TUN/TAP device driver support
`

Is this because of the device I am using RG503 or Arkos does not support this module yet? Can you support it?

<sub>Sent from <a href="https://apps.apple.com/cn/app/id1314212521">PPHub</a></sub>

christianhaitian, 2023-01-12:
Normally, you'd have to compile the kernel and add that module to ArkOS.  The kernel source for ArkOS for the RG503 is available here (https://github.com/christianhaitian/rg503Kernel).  Not something I will add in for general support at this time but you're more than welcome to give it a shot for your personal use.  Since I was already in the kernel source reviewing other items, I went ahead and compiled this particular module.  Just take the tun.ko file that's within the attached zip file and copy it to `/lib/modules/4.19.172/kernel/drivers/net/` then do `sudo depmod`.  Then try your application again.  Good luck!
[tun.zip](https://github.com/christianhaitian/arkos/files/10404219/tun.zip)

closed 2023-01-12 by christianhaitian

jokerwho, 2023-01-15:
Thanks a lot! That fix my problem. lol