christianhaitian/arkos #610: Stuck on last installation boot in RG351P

Kherham, 2023-01-13:
Hello Christian.

First of all, thanks for your work assembling these ROMs. I know you ended releases for RG351P/M but I tried so many times to install ARKOS in my device but ever got the same problem. It takes the first boot after I turn on the device, show all directories and files being installed but in last boot, that on to start system, simply the screen froze and dont finish the installation.

I tried a lot of times and ever got the same problem. Few days ago, after about 20 tries, system finished installation and I could use it but some day later, my sun use my SD Card accidentally and I lost the system. Now am trying to install it again but ever got the same problem.

Device: RG351P
SD Cards used: Original 64gb, SamDisk Ultra 125gb c10 xc (2 cards in first usage)
[image]

Kherham, 2023-01-13:
Dont go ahead of this screen.

christianhaitian, 2023-01-13:
I have no recommendation for this unit.  It could be Anbernic has changed something recently that makes the unit incompatible with this customer firmware.

closed 2023-01-13 by christianhaitian