christianhaitian/arkos #611: RG353M - ArkOS Won’t get Past Splash Screen Without Soft Reset

ddc99999, 2023-01-13:
I pulled the correct RG353M ArkOS image file from the GitHub link provided here. I flashed it onto a new SanDisk 32GB SD Card (bought from a ships and sold by amazon, amazon prime, vendor). SD Card formatted to FAT32 first, then the image loaded to ensure it was clean.

After inserting the SD Card into the RG353M, the device got got hung up after the splash screen (waited a while, it wasn’t installing). I soft reset it and then after the splash screen comes up it proceeds to finish the install like I see in other How-To's and Videos.

I was able to completely set it up from there, even enabled Wi-Fi and did an update check to ensure the ArkOS v2.0 was up to date.

But no matter what, if I turn off the device then when I reboot I need to wait for the splash screen to disappear, then hit reset, then let it power on again properly.

I’m not the only one with this ArkOS issue from my searching. Nobody seems to have a clue other than blaming hardware (SD card, Handheld itself) or saying the image didn’t burn right onto the SD (highly doubtful). I am using a SanDisk 32GB card that I format to FAT32 to ensure it's clean, then load the ARKOS image to. I tried to do it again from scratch on a new card and again, same exact problem.

Any ideas? I’m convinced it’s the ArkOS software not playing well with the device. It's not like I can't get it to work and installed a corrupted file. But what do I know? It works 100% right in all aspects other than needing to be reset once after shutdown.

I’m about to just switch back to the stock software which was running fine. That or go permanent android mode.

Thoughts or tips??? Maybe this was fixed in other Anbernic devices but not the RG353M yet? If I could get my hands on an older image file, I could try that. Also may bite the bullet and try and load ELEC instead and see what happens even though I have zero interest in using that.

IsaacHeron, 2023-01-13:
Hi! I had similar issues with my RG353M. Always required one or two resets to get ArkOS to boot, but also had frequent crashes in EmuStation, Pico-8 (official), both file managers... you get the idea. I also could install uOS or JelOS, both reported some error after booting from the flashed SD card. Stock OS also had some problems. After ordering another SD card from Lear (had tried SanDisk, Toshiba and Transcend in different combinations) I remembered Android. So I booted Android and.... it crashed after a few tabs in the UI. Several times.
Compared to what you are reporting my device was crashier than yours, but the really strange symptoms (a few resets to get it to boot after it was off for some time) are the same. I decided to just order a replacement from Amazon (hell, I'm paying a premium over there anyway) and lo and behold: The new device was able to install uOS, ArkOS doesn't crash anymore and Pico-8 works like a charm. Also the interference noise from the speaker when Wi-Fi is active is gone. I really had a defective unit. I've been testing Pico-8 games (totally missed that one until now), editing a EmuStation theme over smb connection, etc. with just two surprise reboots-in both cases I was dumb and accidentally pressed the reset button...
I'm leaving this here because I, too, was only able to find "advice" in the vein of "SanDisk has problems" and "you've flashed the card wrong".
And to complete my report: the problematic device was in a box as wide as the device itself with the cable under the white inlay the device rests on. The working one came in a slightly wider but slimmer box with the cable in a separate tiny box positioned next to the device and its inlay. Optically and from the markings on the back there is no difference so it might just be coincidence that Anbernic changed the box design and I got one of each with two orders just one week apart.

christianhaitian, 2023-01-13:
I'm running ArkOS on my RG353M without any of these issues using authentic Sandisk and Samsung sd cards.  So I do believe the issue is the hardware.  It may work fine with the stock os for now but you may find it problematic there after awhile.  As Isaac stated, you may find Android unstable after awhile as well.  You can give UnofficialOS a try which is the same OS as JelOS just precompiled since JelOS went source only.  Or you can choose to remain with the stock os since I hear it's not bad there.  Just keep in mind the unit may have further issues in the future once you continue to use it.  Also, make sure to completely charge the unit before installing any new software.  The units typically come with about 70 - 90 percent charge per what the meter shows but not sure how reliable that really is.  good luck.

closed 2023-01-13 by christianhaitian

ddc99999, 2023-01-13:
I haven't had any other problems with the device other than the one listed. No issues with android, no excessive noise, unit looks flawless, etc.

I wouldn't even think there was an issue if not for this custom firmware install.

I did make backups of everything I could, and I'm running android 11 using the software installed directly on the device (not the SD). So far so good. I ordered the unit directly from Anbernic and I'm not going to return it when it works perfect except for the custom software. I also spent so long setting things up I'm not returning it on a hunch from the internet.

Thanks for all the advice though. I think i'm best off though hoping I don't get growing problems like you both mentioned.