christianhaitian/arkos #614: Adding options to the scraper

n-studio, 2023-01-16:
Sorry if I'm asking the question on the wrong repository, I'm confused with the difference between RetroArch/Arkos/EmulationStation.

Is it possible to add the ability to not update the ROM title when scraping?
My ROM titles include valuable information such as `(USA)`, `(rev)`, `(beta)`, etc. that are removed when the ROM metadata is scraped. It would be great if we could choose to display the filename instead of the game title.

In the meanwhile I manually edit the `gamelist.xml` file.

Is it also possible to automatically delete the image/video files of deleted roms?

Thank you for the great work!

christianhaitian, 2023-01-16:
So pressing the start button then going to UI, then scrolling down to Show Filenames in List and turning that on doesn't give you this info?  It should for all except arcade systems such as CPS1,2,&3, Mame, and Arcade.

n-studio, 2023-01-17:
Indeed it does! I'm very sorry I completely missed that option. Thank you!

closed 2023-01-17 by n-studio