christianhaitian/arkos #615: Is it safe to install RG351MP releases on PowKiddy RGB20S ?

RkShaRkz, 2023-01-17:
Not really an issue, more of an general inquiry since the RG351MP's FAQ (https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351MP) says nothing about this, however
> strongly suggests that the PowKiddy RGB20S is a clone of the Anbernic RG351MP, but with a different layout. You'll want the RG351MP version (not RG351M) of ArkOS

So is this credible information, meaning that RG351MP-related release is safe to use on the PowKiddy RGB20S, actually this file (https://drive.google.com/file/d/1IUjqPDJ1CkDZPCuULm0wiqgXBov9l0je/view) which, by the way, imply 07102022 rather than rather than 01152023-1 as infered from here https://github.com/christianhaitian/arkos/releases/tag/v2.0.01.15.2023-1

If i fork that tag or build the code from that release, will it base safe to install on my PowKiddy RGB20S or not? The only one mentioned that it does not work on RGB20S is this one: https://drive.google.com/file/d/1x0Z1LHhMntP6B6V5ePmioxp8vAdlt58Z/view?usp=sharing

### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?
RGB20S

### What version of ArkOS are you on?
11052021

christianhaitian, 2023-01-18:
I do not know.  Try UnofficialOS since they're officially supporting the rg20s.

closed 2023-01-18 by christianhaitian

RkShaRkz, 2023-01-18:
It already runs on ArkOS 2.0 (11052021) hence my question - were there any or many breaking changes since then, or your master branch just diverted and cut off support for RGB20S by accident / on purpose?

@christianhaitian

christianhaitian, 2023-01-18:
Hard to say since I don't know what changes Powkiddy made to ArkOS for the RGB20s.  You can take a chance to update to the latest updates and see what happens.  I would take an image of the card as is before you do so in case you need to revert back.  Good luck.

RkShaRkz, 2023-01-19:
Yes, that's exactly my plan - backup the current microSD cards' images, burn them to new bigger SD cards (mainly because i heard the cards they provide are junk), and then have an open field to try stuff out.

Try burning the latest RG351MP release and see whether it works, or the unofficial OS (since they already did what i planned to do - add PS2 emu support) and see what's there to see.

However, there's one important question - how big are the chances that it fries/bricks the device? I can always revert if it doesn't boot, but i can't revert a fried board.

christianhaitian, 2023-01-19:
I'm not confident about anything related with the rgb20s.  I don't have the unit and again, I did not make any adjustments to account for any particulars about the unit.  I can't say what will or will not happen with the device as I don't know how Powkiddy even provisioned the device.  I just don't know anything about it.  Sorry.

RkShaRkz, 2023-01-20:
After making an image of the OS card and looking at it's `boot.ini`file, i see stuff relevant to what i posted in the opening post.
```
# Boot Arguments
setenv bootargs "root=UUID='e139ce78-9841-40fe-8823-96a304a09859' rootwait rw fsck.repair=yes net.ifnames=0 fbcon=rotate:0 console=/dev/ttyFIQ0 quiet splash plymouth.ignore-serial-consoles consoleblank=0"

# Booting
setenv loadaddr "0x02000000"
setenv initrd_loadaddr "0x01100000"
setenv dtb_loadaddr "0x01f00000"

load mmc 1:1 ${loadaddr} Image
load mmc 1:1 ${initrd_loadaddr} uInitrd

load mmc 1:1 ${dtb_loadaddr} rk3326-rg351mp-linux.dtb

booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}
```

So that gives me a slight nod/push that RG351MP builds of ArkOS might as well be compatible with the device.

However... Unofficial OS (https://github.com/RetroGFX/UnofficialOS/releases) clearly states that it does support the RGB20S. 

That being said, lemme be your betatester / spirit of confidence. Meaning, i'll install the latest RG351MP build you have released, and tell you whether the RGB20S boots nicely with it or not, then you're free to do whatever you want with that information.

So then, instead of the "I just don't know anything about it", you'll be able to say "this one guy tried it and told me it works / doesn't work, so feel free to take his word with a grain of salt" to future inquirers @christianhaitian

> I don't have the unit and again, I did not make any adjustments to account for any particulars about the unit.

this is another positive nod in the direction that it can/could work.

CrazyXacker, 2023-01-26:
Installed ArkOS xx012023 (RG531MP) into RGB20S. Works perfectly. And UnofficialOS/JELOS doesn't even boot despite mentioning support