christianhaitian/arkos #616: Bios directory not pointing to SD2 after switching to SD2 for roms.

kondor505, 2023-01-18:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Possible problem of retroarch bios directory not pointing to roms2/bios after switching to SD2.  

### How can the issue be reproduced? (If applicable)
clean install then SWITCH TO SD2 FOR ROMS

### What device are you using?
rg353v

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
01142023

### Anything else you'd like to include that may help to review this issue or feature request?
 Possible to add to FAQ informing that user might have to point retroarch to roms2/bios  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG353V#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games

christianhaitian, 2023-01-18:
Switched just fine on my end with both retroarch and retroarch32 using the latest available image and from a previous image that was updated to the latest ota udpates.

kondor505, 2023-01-18:
Must be my end because I am also seeing portmaster pointing to SD1 for roms which means they do not show after install and restart, I am wondering if I restored retroarch default settings and that messed it up for me.  All good, I am learning this OS.  Thanks for the hard work.  Closing.

closed 2023-01-18 by kondor505

christianhaitian, 2023-01-18:
oh yea, if you restored your settings from another device that's supported by ArkOS, that will cause an issue.  If you restored your settings from the same device but it used just 1 sd instead of 2 and that will cause an oddity but usually, doing switch to main sd, reboot, then switch back to sd2 should take care of the issue.